SpenceKonde/megaTinyCore #2: Reorganize files so you can sync it to hardware folder

SpenceKonde, 2019-06-03:
I'm going to rearrange the files so it can be sync'ed to hardware for manual install, instead of being board manager only. 

Also need to update platform.txt to identify it at megaTinyCore. Once things are working, we'll probably want to remove the non attiny boards. I'm going to look into the available megaAVR attiny parts and order some in SOIC to start playing with this. 

What kind of programmer do I want to buy for these, to get started?

freemovers, 2019-06-03:
I have been using the solution from [https://github.com/ElTangas/jtag2updi/tree/master/tools/avrjtagicev2](https://github.com/ElTangas/jtag2updi/tree/master/tools/avrjtagicev2) for uploading the sketches.
I forgot to mention that avr-gcc 7.3.0 and the MEGA DFP are required , but that will automatically install when you install the megaAVR boards in Arduino first.

freemovers, 2019-06-06:
I changed the folder and file structure on the fork I created. Will it be more efficient to work directly from the SpenceKonde instead?
Simon was able to program the ATtiny from the Arduino Nano:
[https://hackaday.io/project/165439-attiny-0-series-programming-on-the-cheap](https://hackaday.io/project/165439-attiny-0-series-programming-on-the-cheap)

SpenceKonde, 2019-06-06:
Yeah, let's get that change into my repo, when I get home I'll try to figure out how to make you a collaborator so you can push changes without me needing to approve them - at least until this is ready for first release. 

I sent you my mailing address for the board you made; right now since I don't have hardware I can't test any of this stuff - did you get that email? I'm planning to place a digikey order this weekend and pick up a bunch of megaavr ATTiny's of varying types to start making new variants.

SpenceKonde, 2019-06-06:
Actually, just figured out how to do collaborator, if you accept the invite you should be able to push to this repo too. That should make development smoother

freemovers, 2019-06-06:
Excellent, I updated my hardware folder to match your repository as well so that we do not have multiple forks going at this point of the development. I did sent some boards (ATtiny1616 only) out to you on Tuesday from the West Coast by snail mail, so you might not see them till the beginning of next week.

SpenceKonde, 2019-06-06:
Wonderful, I look forward to getting them. Combined with the stash of new parts I'll have coming, I look forward to being able to really dig into these new parts.

SpenceKonde, 2019-06-17:
This is long since done.

closed 2019-06-17 by SpenceKonde