SpenceKonde/megaTinyCore #5: API missing

freemovers, 2019-06-06:
The original Arduino core has a separate repository for the API. Add installation instructions for the API as well (see following note from the readme file), or add the API folder to this repository:

> **Developing**
> 
>     Create an <SKETCHBOOK>/hardware/arduino folder. Where <SKETCHBOOK> is the location of your Arduino sketchbook.
>     Change directories: cd <SKETCHBOOK>/hardware/arduino
>     Clone this repo: git clone https://github.com/arduino/ArduinoCore-megaavr.git megaavr
>     Change directories: cd megaavr/cores/arduino
>     Copy or symlink the api folder from the ArduinoCore-API repo.
>     Restart the IDE

SpenceKonde, 2019-06-06:
Do you have any background info on this you can link? I'm confused as to what is included in the "API" and what isn't?

freemovers, 2019-06-06:
After deleting my local version to link it this repository I noticed this error and found the following note in .gitignore:

> Adding the folder "avr/cores/arduino/api" to ignore list
> this folder point to the repo below, and note not "master" but "api-release-cpp"
> https://github.com/arduino/ArduinoCore-api/tree/api-release-cpp
> api

See the note and link under the developing section of the following readme file:
https://github.com/arduino/ArduinoCore-megaavr/blob/master/README.md

SpenceKonde, 2019-06-06:
Oh for fucks sake. 

I vote we pull the API stuff in so the core stands on it's own, but I think we should get Per's input.

freemovers, 2019-06-06:
LOL, this is what Per said back in February:

> Note that Arduino megaAVR Boards uses the new ArduinoCore-API system. All the non-architecture specific core code has been moved to a separate repository, from which it can be shared between all cores. That means less code you need to worry about porting and maintaining. I'd love to see this project be one of the first 3rd party hardware packages to take advantage of it. You should be able to use a subtree for this.

I'm all for maintaining less code and keep it compatible for most devices out there when possible.

per1234, 2019-06-06:
Some background on the ArduinoCore-API repository:
The idea is to move all the non-architecture specific code of the core library into a separate repository from which it can be shared by all cores (official and 3rd party). This has the following benefits:
1. Reduces the work required to create a new core because you only need to consider possible changes to the remaining core library files.
2. Makes it easier to keep core libraries updated. You only need to pull in the latest release of the ArduinoCore-API repository before doing a release.
3. Makes it less likely for cores to break compatibility with the standard Arduino core API (a perfect example being the mess created by the ESP folks deciding to [do their own thing with the Server class](https://github.com/espressif/arduino-esp32/commit/5d158505a120c84df50092fb4525e45de79646b8), breaking ESP32 compatibility with all the external networking libraries).

I would really like to see the 3rd party core developers get on board with this initiative. So far, I'm only aware of one 3rd party core that is using ArduinoCore-API.

My advice would be to add the ArduinoCore-API content directly to this repository so that people can easily do beta testing by simply downloading the megaTinyCore repo and unzipping it to their `hardware` folder. If you want to make it so the updates from ArduinoCore-API can be done purely via Git commands, you can add it as a subtree in the `megaavr/cores/arduino/api` folder

SpenceKonde, 2019-06-07:
Had to read that a few times to follow it.... 

I think we do want to put the API content directly into this repo, and periodically update it - I think it is very valuable to be able to sync the repo straight from github to do a manual install of the github version.

freemovers, 2019-06-07:
I agree, we can always make it dynamic to Per's suggestions when we get to a release.

closed 2019-06-07 by freemovers