SpenceKonde/megaTinyCore #6: Discussion: Is this the right pin mapping?

SpenceKonde, 2019-06-07:
Before people actually start using this stuff we should make sure that we've picked a good pin mapping - it's absolute hell to realize later that your pin mapping could have been better, and have to go back and change it (leaving people who are using the core out in the cold), or offer a second pin mapping (confusing new users). Best to put some thought into this now. 

The current pin mapping has the advantage of matching the SOIC pinout, so the numbering on a PCB will be continuous. . 

On the downside, pins on port A are split between the lowest and highest ports, and most of the pins are numbered in the opposite order as their bit positions, and the analog pin numbers don't match the digital ones, which is a nice feature to have. 

Pity they picked PA0 for reset - PB5 (the last bit in a weird 6-bit port) would have been the natural choice IMO.

freemovers, 2019-06-07:
I agree, here are some more questions open for debate:

1. Assign pin#17 to PA0 in case someone wants that extra pin for I/O (I agree on your UDPI / Reset, they could have used a different port for that.
2. Use ADC1 for the AIN (starts at PA4 with AIN0, so is cleaner)
3. PA6 is DAC, should that be A0 like the M0 boards?
4. Should PORTMUX for SPI, TWI and UART be flexible through an option on the menu?

You might have noticed the PORTMUX talbe and pin layout in the pins_arduino.h, see below for reference as well:

```
                          _____ 
                  VDD   1|*    |20  GND
 (nSS)  (AIN4) PA4  0~  2|     |19  16~ PA3 (AIN3)(EXTCLK)
        (AIN5) PA5  1~  3|     |18  15  PA2 (AIN2)(MISO)
 (DAC)  (AIN6) PA6  2   4|     |17  14  PA1 (AIN1)(MOSI)
        (AIN7) PA7  3   5|     |16      PA0 (nRESET/UPDI)
        (AIN8) PB5  4   6|     |15  13  PC3
        (AIN9) PB4  5   7|     |14  12  PC2 
 (RXD) (TOSC1) PB3  6   8|     |13  11  PC1
 (TXD) (TOSC2) PB2  7~  9|     |12  10  PC0
 (SDA) (AIN10) PB1  8~ 10|_____|11   9~ PB0 (AIN11)(SCL)
```

SpenceKonde, 2019-06-18:
I added a proposal for pin mapping on the -4 series

SpenceKonde, 2019-06-18:
Re: 3, I think the DAC should have the numbering as you'd expect from the pin location, not A0 or something. We can either make analogWrite() use the DAC when called on that pin, or create a builtin library (say DAC.h) which you'd #include and would give a DACwrite() function. 

Re: 1 Having looked at the pinouts for the chips with a different number of pins, the PA0/UPDI pin decision looks a little less stupid, but the fact that it still blocks up the only full 8-bit port is still lame

SpenceKonde, 2019-06-20:
I don't see any convincing reason not to use this pinout, and the corresponding ones for the other series' 

I will submit variant files tonight.

closed 2019-06-21 by SpenceKonde