SpenceKonde/megaTinyCore #10: Support t816, t416

SpenceKonde, 2019-06-16:
There are a few subtle differences between the t1616/3216 and t416/816. We need to account for these. At present it's not clear to me whether this can be done with the same variants files or whether a new variant is needed. But it looks like it may just work? I don't see TCB being used for anything?

freemovers, 2019-06-16:
TCB is used for millis, but it is defined with the extra flags in the boards.txt:
`at1616.build.extra_flags=-DMILLIS_USE_TIMERB1 -DNO_EXTERNAL_I2C_PULLUP`
Could make that part of a menu option, or move the millis to TCB0 in general.

SpenceKonde, 2019-06-16:
Moved millis to timerB0. 

We needed to do that anyway because tone() uses timerB1 so tone() would hose millis() if it used timerB1.

closed 2019-06-16 by SpenceKonde