SpenceKonde/megaTinyCore #13: Clock speed menu option

SpenceKonde, 2019-06-16:
Looks like our options are 1MHz internal, 16MHz internal, 20MHz internal?

freemovers, 2019-06-16:
The are 3 oscillators to pick from:
OSC20M (20MHz)
OSC16M (16MHz)
OSCULP32K (32KHz (Ultra Low Power))
With one of these 3 oscillator the main clock (for the CPU) can be configured. Using it at 16/20MHz, or using a prescaler with the following options:
2, 4, 8, 16, 32, 64 and some odd prescalers like 6, 10, 12, 24, 48

SpenceKonde, 2019-06-18:
Okay, so we have 16MHz, 20MHz - and we're going to need some lower speeds too.

freemovers, 2019-06-18:
Use the 16/20MHz with all the available pre-scalers will result in 23 different possible main clock speed. The pre-scaler is currently calculated in:
https://github.com/SpenceKonde/megaTinyCore/blob/acc780991b663587c3446d73d247bc418c66abc3/megaavr/cores/arduino/wiring.c#L291
Should we offer main clock speed offerings and make it match the current pre-scaler option in wire.c, or have the pre-scaler defined by a menu option in boards.txt. For example someone picks a clock speed of 1.25MHz, OSC20M will be used, prescaler division of 16, f_cpu set to 1.25MHz. 

OSCCFG   | Prescaler   |  Main Clock
-- | -- | --
20000000 | 1 | 20000000
16000000 | 1 | 16000000
20000000 | 2 | 10000000
16000000 | 2 | 8000000
20000000 | 4 | 5000000
16000000 | 4 | 4000000
20000000 | 6 | 3333333
16000000 | 6 | 2666667
20000000 | 8 | 2500000
20000000 | 10 | 2000000
16000000 | 8 | 2000000
20000000 | 12 | 1666667
16000000 | 10 | 1600000
16000000 | 12 | 1333333
20000000 | 16 | 1250000
16000000 | 16 | 1000000
20000000 | 24 | 833333.3
16000000 | 24 | 666666.7
20000000 | 32 | 625000
16000000 | 32 | 500000
20000000 | 48 | 416666.7
16000000 | 48 | 333333.3
20000000 | 64 | 312500
16000000 | 64 | 250000

freemovers, 2019-06-18:
And what if someone wants to use the OSCULP32K. In my opinion that should not be possible. Only reason for running it that slow could be for power consumption, but in that case they should use the different sleep modes to turn off the OSC16M or OSC20M entirely.

SpenceKonde, 2019-06-21:
I agree on the 32Khz ULP - won't that also cause problems with the UPDI programming? Don't even get me started on people requesting stupidly slow clock rates because they think that it will reduce power consumption in sleep mode when the cpu clock is off. 

The speed grades we need are:
Something no greater than 5MHz (for 1.8v)
Something no greater than 10MHz (for 2.7v - in practice 3.3v)
20MHz

I would vote 20MHz, 16MHz, 10MHz, 8MHz, 5MHz, 4MHz, 1MHz. 

We will need to make sure that millis, micros, delay and delaymicros work correctly at these clock speeds (for that matter, we also need to test all of those at 20MHz and make sure they are right, or close to it; the default implementation of some of them on the official avr classic core was wrong, and there was much hemming and hawing and implementation effort to get it sorted in attinycore. Luckily since we don't have external crystal support here, we don't need as many weirdo frequencies supported (I wound up hand-tuning some of the functions with an implementation using bitshift operations, because division was too slow). 

Which reminds me - what mathematical operations are handled in hardware and which ones in software on the megaAVR tinies? classic atmega has hardware multiply, but software division, so division crawls but multiplication doesn't. classic attiny doesn't have hardware multiply, so multiplication is somewhat slow. 

What is the maximum UPDI baud rate relative to clock speed? We may need to adapt the UPDI programmer sketch if the baud rate it defaults to is too high.

freemovers, 2019-06-21:
I did run the Main clock on 32Khz  before (just because you can) and had no issues with the UPDI. I think UPDI (like the TCD) is asynchronous from the main clock.
If someone wants to run another frequency they can always set the register for the divider in the code, and let them figure out delay and millis.

SpenceKonde, 2019-06-21:
Huh! Well that's cool. I still think it's way too slow for any practical purpose, so IMO we shouldn't do it.

commit 876f25b9 (2019-06-22): Initialize prescalers

For supported clock speeds of 20, 16, 10, 8, 5, 4, 1 MHz - fix #13
closed 2019-06-22 by SpenceKonde