SpenceKonde/megaTinyCore #14: DAC not implemented

freemovers, 2019-06-16:
Every ATtiny 1-series supports DAC to write to PA6 for a real analog output. analogWrite only supports PWM at this moment. Should we use analogWrite for the DAC (like it is done on the SAMD) or use something like dacWrite (like the ESP32).

SpenceKonde, 2019-06-21:
I think this should either be analogWrite(), or we should have a library to include that adds a dacWrite() - we shouldn't have non-API user-facing functions in the core.

SpenceKonde, 2019-06-21:
I think we need to do it with analogWrite, because turning off the DAC means adjusting turnOffPWM in wiring_digital.c? 
analogWrite route would mean
add a DAC define to Arduino.h, (like the timers)
add #ifdef to pins_arduino timer list for the DAC pin that uses that DAC define
add case to wiring_analog's analogWrite(), treating it like a timer
add case to wiring_digital's turnOffPWM() treating it like a timer
adjust the pwm pin macro to say it has pwm since we're treating it like PWM.

SpenceKonde, 2019-06-26:
Uuuugh! Are you fucking shitting me?! 

You have to use a VREF voltage for the DAC. You cannot use Vcc. So there's another menu on boards.txt just to set the bloody vref voltage.

commit 93866f91 (2019-06-26): DAC support (basic) 

Should work now!  (fixed #14) - but see also #55.
closed 2019-06-26 by SpenceKonde