SpenceKonde/megaTinyCore #16: Pick a timer, any timer!

SpenceKonde, 2019-06-16:
Important decisions need to be made regarding which timers to use for what on which chips.

freemovers, 2019-06-17:
Here are my 5 cents:
TCA0   PWM
TCB0   Millis
TCB1   N/A for all chips (tone?)
TCD    Servo's (not available on 0-series)
RTC     timer interrupts (low power applications)
How about using the RTC for millis and delay, and placing CPU in idle when delay is used? For low power applications someone could just change the sleep mode to Standby, and conserve energy by using delay without changing the code.

SpenceKonde, 2019-06-17:
Tiny 0-series doesn't have TCD. 
Smaller 1-series don't have a second TCB.

I would love to use the RTC timer for millis - but there's a problem. There's also micros() and you can't clock the RTC timer off the system clock as far as I can tell, so you couldn't make it give microseconds.

SpenceKonde, 2019-06-17:
And as noted in separate issue, we are not using TCA0 correctly in split mode for PWM - only half of the outputs work.

SpenceKonde, 2019-06-17:
I'd love to cut micros() loose, use the RTC for millis, and so on, but what then would we do for micros()

SpenceKonde, 2019-06-17:
Beyond whining about how nice it would be if we didn't have to be compatible with the arduino API, there's indeed a puzzle about how to distribute the existing TCBs and TCD.... 

I kind of like the idea of stuffing millis onto TCD, except then there's the whole 0 series.

SpenceKonde, 2019-06-17:
I wonder if there's a way that TCA could be used for millis as well as PWM - will study the datasheet. If it is possible, then that would greatly simplify things

SpenceKonde, 2019-06-19:
Was that all to supporting TCA millis?!

Note - it will not work without TCAsplit.

SpenceKonde, 2019-06-19:
We can get 2 additional channels of PWM on many of the 1-series with TCD on PC0 and PC1! 8 PWM channels on chips that have those pins!

TCBn is a no-go for additional PWM because the output channels are also usable via TCA.

SpenceKonde, 2019-06-21:
Okay, so I think we're done here?

closed 2019-06-21 by SpenceKonde