SpenceKonde/megaTinyCore #8: analogWrite() does not work on pins 0 and 1

SpenceKonde, 2019-06-16:
Arduino pins 0 and 1 (PA4,5) are listed in pins_arduino as supporting PWM. analogWrite does nothing on these pins.

SpenceKonde, 2019-06-16:
Also broken on pin 16 (PA3). 

Core seems to be treating these pins as though they don't have PWM support - writing values below 128 writes them LOW, values above 128 write them HIGH.

SpenceKonde, 2019-06-16:
Maybe the issue is in the configuration of TCA? Looking at the datasheet it looks like you can only get 6 PWM channels out of it if it is in "split mode". Is that configured correctly? My guess is it is not.

freemovers, 2019-06-17:
We must rewrite the analogWrite function to split the compare counter, here is an example:
https://github.com/MicrochipTech/TB3217_Getting_Started_with_TCA/blob/master/Generating_two_PWM_signals_in_split_mode/main.c

SpenceKonde, 2019-06-18:
I think this will be the next thing I work on, since it sidesteps all the design decisions (while I'm at it, I'll see if I can put millis() on this timer? (#16) It looks possible...)

SpenceKonde, 2019-06-19:
Oh god...
`uint8_t bit_pos  = digitalPinToBitPosition(pin);`

I was wondering how they made the timers work without specifying the channel over in the variants file....

SpenceKonde, 2019-06-19:
I think the solution in SplitTCA branch should make this work, but can't test until I get home.

SpenceKonde, 2019-06-20:
This works now \o/

closed 2019-06-20 by SpenceKonde

freemovers, 2019-06-20:
This is awesome! Very cool that millis is working on TCA as well. Following the steps you took to get this implemented taught me more about these timers as well. Thanks!