SpenceKonde/megaTinyCore #9: Suspect entries in boards.txt

SpenceKonde, 2019-06-16:
Why is there a vid/pid here? These boards to not have native USB, and are used with a jtag2updi programmer. 

at1616.vid.0=0x2341
at1616.pid.0=0x0058

Why is there a bootloader file for the 4809 when these are programmed via UPDI without a bootloader? 

Is that fuse file correct for the tiny series? Is "burn bootloader" required for the new megaAVRs to set fuses to the correct clock speed or other parameters? 

at1616.bootloader.tool=avrdude
at1616.bootloader.file=atmega4809_uart_bl.hex
at1616.bootloader.SYSCFG0=0xC9
at1616.bootloader.BOOTEND=0x00
at1616.bootloader.OSCCFG=0x01
at1616.fuses.file=fuses_4809.bin

freemovers, 2019-06-16:
Most of these were copied over from the Arduino Nano Every. vid and pid are not needed for sure. Fuse settings are listed on page 32 in the datasheet, but you can change most of these within the code (clock speed is all done in the code). SYSCFG0 might have some interesting fuses like the EESAVE to save EEPROM data during chip erase. Are there plans for supporting Optiboot for these as well, because then we should be able to set the  the end of the boot section with the BOOTEND register.

SpenceKonde, 2019-06-16:
Support for a bootloader will happen in the distant future, let's not worry about it for now.

SpenceKonde, 2019-06-16:
Okay, only questions now is the fuses file.

SpenceKonde, 2019-06-17:
It's not even referenced in platform.txt! Out it goes.

closed 2019-06-17 by SpenceKonde