SpenceKonde/megaTinyCore #21: Support 3217/1617/817/417

SpenceKonde, 2019-06-18:
As with before, should be reasonably simple, just a pinmapping. but complicated by the fact that the parts come only in awful, inconvenient packages for prototyping.

per1234, 2019-06-18:
I see Microchip sells an [ATtiny3217 Xplained Pro Evaluation Kit](http://www.microchip.com/Developmenttools/ProductDetails.aspx?PartNO=ATTINY3217-XPRO) and an [ATtiny817 Xplained Mini](http://www.microchip.com/Developmenttools/ProductDetails.aspx?PartNO=ATTINY817-XMINI). I'm happy to donate some of those to the cause if it would be helpful to the project.

SpenceKonde, 2019-06-21:
This is done (of course, nothing tested) 

I have an 817 xplained mini kicking around from that promo on arrow electronics, but haven't powered it up or anything :-P

closed 2019-06-21 by SpenceKonde