SpenceKonde/megaTinyCore #23: Add Servo library

SpenceKonde, 2019-06-18:
We need a megaAVR compatible implementation of Servo.h. See also #16 regarding where to use what timer.

SpenceKonde, 2019-06-21:
As for what timer - we should use TCB1 if it's available, otherwise TCB0 (and lose Tone)

freemovers, 2019-06-21:
Looks like the servo library is already updated for the MEGAAVR (but not standard in the MEGAAVR core):
https://github.com/arduino-libraries/Servo/tree/master/src/megaavr

SpenceKonde, 2019-06-21:
Then we need to pull in a copy of it and strip out non-megaavr stuff and put it into the libraries folder, as it needs to be adjusted to work on tiny

SpenceKonde, 2019-07-01:
I threw the servo library linked above in, and set it up to use the timers I said it would use. It should now work, assuming the original library worked.

closed 2019-07-06 by SpenceKonde