SpenceKonde/megaTinyCore #24: Pinout diagrams

SpenceKonde, 2019-06-18:
This is @mcudude's department, but I'm going to need his help here. Of course, this is gated by the actual creation of said pinouts!

freemovers, 2019-06-20:
[image]
Will this work. Since we no longer have A# it is showing digital, or digital/analog by color. Not sure about the PWM outputs yet ... Any feedback?

freemovers, 2019-06-20:
Okay, pin#8 next to PB0 should be 9!

freemovers, 2019-06-20:
Here they are. It is actually not bad since they are all so similar:

[image]

MCUdude, 2019-06-20:
I become a dad last week, so my time is kinda limited. As for now you only need pinouts for ATtiny 406/806/1606 and ATtiny416/816/1616/3216? Do you want "dedicated" QFN style pinouts too?

And what are the plans on the other tinyAVR-0 and tinyAVR-1 families? These diagrams seems to be the most recent ones.
[image]
[image]

I have development boards for ATtiny416, Attiny1616 and ATtiny104, so I'm able to help out with testing if I have some spare time.

SpenceKonde, 2019-06-20:
It looks like @freemovers is able to generate pinout images, so you may be off the hook (grats on the kid!)

The x7 image above depicts a package that does not exist - it's only available in qfn. So we need an image based on that. I have a few other thoughts on things to include in the images that I will post a bit later

MCUdude, 2019-06-20:
> It looks like @freemovers is able to generate pinout images, so you may be off the hook

Yes, they will definitely do the job! The only thing is that it's very hard to figure out what family ATtinyX(X)X6 really is. Oh, I just found out that ATtiny104 as nothing in common with the ATtiny204/404/804/1604 family.

>  (grats on the kid!)

Thank you! I'm pretty sure I will have to prioritize my a little differently, but I still want to be active here on Github. Time will tell, but if I may be a little slow to reply you now know the reason 😉

freemovers, 2019-06-21:
Here is the QFN version:
[image]
I agree that the naming is confusing, but it was more for me to keep track of the differences. It is probably good to make separate pages for the 0-series or 1-series. @SpenceKonde do you want to use the Wiki to make separate pages for each chip? Maybe we can use the Figure MCUdude posted and link each chip it to a page with the pin layout and other specifications (from the spreadsheet).

SpenceKonde, 2019-06-22:
The wiki? What wiki? 

My plan was to do what was done on ATTinyCore - https://github.com/SpenceKonde/ATTinyCore/tree/master/avr/extras

My thinking was to create extras folder with 8 documentation files in it:
ATtiny_x16.md
ATtiny_x17.md
ATtiny_x14.md
ATtiny_x12.md
ATtiny_x06.md
ATtiny_x07.md
ATtiny_x04.md
ATtiny_x02.md

which would have feature tables and lists of chip-specific caveats. 

We should have a separate pinout image for 0-series and 1-series anyway, since 1-series have DAC, and on the x16 and x17, two additional PWM pins (PC0, PC1). 

x7 image is missing a few connecting lines. 
x2 image is missing TXD, RXD, SCL, SDA markings.

SpenceKonde, 2019-06-22:
pinout diagrams should go in megaavr/extras (in the repo as binaries)

freemovers, 2019-06-22:
@SpenceKonde https://github.com/SpenceKonde/megaTinyCore/wiki
Can you be more specific on the connection lines for the x7, looks to me they are all connected but I might miss something there.
For the x2, are TXD, RXD, SCL and SDA only available when alternative PORTMUX is used, and should PORTMUX be a menu option of done through the sketch?
[image]

SpenceKonde, 2019-06-22:
In the pins_arduino for the x2 series, I do specify the alternate portmux options. I dont quite know how they are used elsewhere in the core.

I dont ser a need to add portmux menu options.if someone needs to change it, that's easy enough to do in the sketch. The way the functions on the pins are arranged in general looks like people wont oftwn have cause to change the portmux settings.

SpenceKonde, 2019-06-23:
Oh! Fantastic! That looks wonderful

The TCD0 PWM pins aren't shown on x16 x17 pinout images, but everything else looks great.

freemovers, 2019-06-23:
Thanks! I just fixed the PWM for the x16 and x17 as well.

closed 2019-06-23 by freemovers

MCUdude, 2019-06-23:
In the pinout diagrams, wouldn't it be better to call the SPI clock *SCK* instead of *CLK*?

SpenceKonde, 2019-06-23:
Good catch, yeah the SPI clock is SCK, not CLK

freemovers, 2019-06-23:
I think CLK slipped in there since the original pinout from the datasheet has EXTCLK (external clock). Updated the diagrams to show SCK now.