SpenceKonde/megaTinyCore #25: Documentation

SpenceKonde, 2019-06-18:
Note that many of these depend on design decisions not yet made - this is mostly to gather up a list of what needs to be done

We need
ReadMe.md with general information
Installation.md - covering manual and board manager installation
Part-specific documentation files - maybe? Maybe not, as these are far less individual than the classic AVR attiny series. Maybe it just needs to be a 0-series and 1-series doc?

SpenceKonde, 2019-06-22:
Also need an idiot's guide to building a jtag2updi programmer from nano/micro/promini

freemovers, 2019-06-22:
I have been using the UNO and a modified JTAG ICE, but is all works the same. I think I have some nano's laying and will take some pictures with those as well.

freemovers, 2019-06-24:
@SpenceKonde, are you going to use the same board manager used for the current ATtiny:
http://drazzy.com/package_drazzy.com_index.json
I did create the jtag2updi programmer instructions for the nano, is that clear, or should it have some pictures?

SpenceKonde, 2019-06-24:
Yes, it will be the same board manager file. We need @per1234 to wave the wand to create board manager releases though - I don't know how to generate the json file (particularly the hashes of the files). What I really want to do is have a script that will generate the board manager file - that could be done if I was better at shell scripting, and had studied the format that the json files use more, and knew how to generate the hashes.

SpenceKonde, 2019-06-24:
re: UPDI programmer, If you think those instructions are going to work, you have far too much faith in users.... going to work over that guide currently (I don't feel like taking pictures, but I think a wiring diagram would be good

So you don't need to put the 10uF cap on them to disable autoreset? I had assumed you did and did it on mine without thinking about it.

freemovers, 2019-06-24:
I started off with the JTAG ICE clone so that I didn't had to deal with the autoreset till SimonMerrett told me he was using the Nano with just the resistor. I have used the UNO and Nano since then without issues, but I did notice the Nano being very slow to start AVRDUDE (so maybe it is resetting a couple of time?). 
SimonMerrett has some good instructions on Hackaday and maybe he is willing to share:
https://hackaday.io/project/165439-attiny-0-series-programming-on-the-cheap/log/163143-programme-your-programmer
I will contact him.

SpenceKonde, 2019-06-24:
I don't think there's a need for bringing in outside experts; I think we're up to the task :-P

SpenceKonde, 2019-06-24:
I updated the updi programmer instructions. We should take some pictures (I'll take one of my perm wired pro mini)

Maybe show the two connection schemes, ie, with a resistor between pin 9 and UPDI pin of a bare tiny, and without the resistor to a UPDI-programming header of a tiny with the resistor on the board. 

[image]

SimonMerrett, 2019-06-24:
Haha! I'm certainly not an expert! Happy to help if I can.

SpenceKonde, 2019-06-24:
If you want to take a stab at making it more readable/understandable for people, I'd welcome the effort - experience has shown that I'm not terribly good at writing instructions for people - I'm too deep into it to know how less-experienced hobbyists think of things, and what they will and won't understand.

SpenceKonde, 2019-06-24:
Made some updates to README.md

freemovers, 2019-06-24:
@SpenceKonde, you mentioned above and in the instructions that you are using pin#9 on the pro mini. On the Nano and UNO PD6 is on pin#6, and from what I can see from the schematics for the pro it should be on pin#6 as well. Can you verify that you use  pin#9?

SpenceKonde, 2019-06-24:
Oops. Yeah, pin 6.

freemovers, 2019-06-24:
What will be in Wiring.md, how to wire up the ATtiny?

freemovers, 2019-06-24:
Great, found an example in the ATtinyCore! Will take out what is not needed on those and get that going.

freemovers, 2019-06-25:
> re: UPDI programmer, If you think those instructions are going to work, you have far too much faith in users.... going to work over that guide currently (I don't feel like taking pictures, but I [...]
After adding the capacitor on the reset pin AVRDUDE does not "wait" after sending the reset and is way faster now. So the 10uF capacitor is definitely recommended.

SpenceKonde, 2019-06-25:
Basic documentation is done, 1.0.0 is released. We of course need to keep improving this... It stays on the list for 1.0.1

SimonMerrett, 2019-06-26:
[image]

I know it's not a schematic but it might be helpful to some people. Have I missed anything - what could be added for clarity (GND and VCC labels on the target? Target and programmer labels?)?

freemovers, 2019-06-26:
I like those pictures, I think for a beginner it is easier to read then a schematic. Looks good, but the ground for the ATtiny40 is currently going to 5V, not GND on the breadboard.

SimonMerrett, 2019-06-26:
[image]
Whoops! Fixed now.

SpenceKonde, 2019-07-04:
Removed from milestones, as "general documentation improvements" is an ongoing process.

SpenceKonde, 2019-07-15:
I'm going to close this, we've done everything on this list. If you think of other things that should be documented, create an issue.

closed 2019-09-03 by SpenceKonde