SpenceKonde/megaTinyCore #27: PIN_An defines don't make sense in variants/txy6/pins_arduino.h

SpenceKonde, 2019-06-18:
What are these PIN_An numbers supposed to mean? 
The last few are correlated with digital pins, but the first few are not. 

@freemovers can you let me know what the intent here was? (and either fix or reassign)

freemovers, 2019-06-18:
These are used on the ATMEGA4809 Arduino board to assign the An to analog inputs. The actual pin# is defined above, so A0 is pin 0 (AIN4), A10 is pin16 (AIN3) . On the Nano Every, all the analog inputs have labels A0 to A7, and the digital I/O is labeled D2 to D13, and there is TX1 and RX0. I just followed that same scheme, but it is very confusing. I rather just call-out the analogRead with the pin# instead of the A#. If you agree we will just eliminate the A# for analog inputs since it is not used anywhere else.

SpenceKonde, 2019-06-18:
I like that idea. If it's not used elsewhere, we can always add it back in if we decide we need a way to use the analog channel number.

closed 2019-06-19 by freemovers