SpenceKonde/megaTinyCore #33: code styling

SpenceKonde, 2019-06-19:
We need to run automated tools over the code to do the following:
* Replace tabs with 2-spaces
* Ensure trailing newline on all files
* Ensure no trailing whitespace on any line

This is required when we add automated travis testing.

per1234, 2019-06-19:
Note that, in my standard CI configuration as used by ATTinyCore, those requirements are imposed on all files, not only the code files. I could configure it to check only code files if that was considered desirable, but I think it makes sense to do it on all files (excluding external resources like ArduinoCore-API).

I recently added a [`check_code_formatting`](https://github.com/per1234/arduino-ci-script#check_code_formatting-strictness-excludedpathlist-targetpath) function to [my CI script](https://github.com/per1234/arduino-ci-script). It can be set to any of three strictness levels. Each level adds additional requirements to the one it and is 100% compatible with the levels below:
1. Code formatting configuration used by the Arduino IDE's **Tools > Auto Format**: https://github.com/arduino/Arduino/blob/master/build/shared/lib/formatter.conf
2. Code formatting configuration used by Arduino on their example sketches: https://github.com/arduino/Arduino/blob/master/build/shared/examples_formatter.conf
3. Custom code formatting configuration I wrote based on common practices in the official Arduino code base: https://github.com/arduino/ArduinoCore-avr/issues/71#issuecomment-466763471

Those configurations are written for the [Artistic Style code formatter tool](http://astyle.sourceforge.net/) used by the Arduino IDE's **Tools > Auto Format**. That tool may be used by itself from the command line or a script to automatically format code. This command recursively formats all code files under the current folder:
```
astyle --suffix=none --options=examples_formatter.conf --recursive  ./*.pde,*.ino,*.h,*.hpp,*.hh,*.hxx,*.h++,*.cpp,*.cc,*.cxx,*.c++,*.cp,*.c,*.ipp,*.ii,*.ixx,*.inl,*.tpp,*.txx,*.tpl
```
You would need to adjust the `--options` option to point to the configuration file you're using.

I'd be happy to submit a PR to bring the existing file in line with the standard. I just need to know:
- Should the generic file formatting checks (https://github.com/SpenceKonde/ATTinyCore/blob/master/.travis.yml#L486-L498) be applied to all files, or only code files?
- Do you want to use the Artistic Style formatting? If so, which formatting configuration?

SpenceKonde, 2019-06-19:
Agree that it should be applied to all files except arduino API and anything else obvious that I'm forgetting. 

Let's hold off on artistic code styling. 

One tricky thing about this PR is that I'm committing changes at the rate of several dozen per day, and it would be really easy for one of those commits to change the code out from under you between your pulling the code and your submitting the PR. I'd say you want to wait until I get home and make sure the split mode branch compiles and merge it (eta 3 hours or so from now), and shoot off an email when you pull the core to make these changes, so I can stop making changes until I see and merge the PR. 

At this stage of core development, a "does it compile bare minimum on every board in boards.txt" automated test alone would be worth it's weight in gold, so I could quickly find when code I submitted busted things on some variant (especially once I add the other 3 sizes of 0 and 1-series parts), so I'd like to get this travising sooner rather than later.

per1234, 2019-06-19:
>wait until I get home and make sure the split mode branch compiles and merge it (eta 3 hours or so from now), and shoot off an email when you pull the core to make these changes, so I can stop making changes until I see and merge the PR.

OK, will do.

>a "does it compile bare minimum on every board in boards.txt" automated test alone would be worth it's weight in gold

I'm happy to set that up. I have listed the possible Travis CI services I can set up in https://github.com/SpenceKonde/megaTinyCore/issues/36, so just comment in that issue thread to tell me which of those you want and I'll get on it.

SpenceKonde, 2019-06-19:
Great, thanks as always. I merged the branch, you're go to create this PR.

SpenceKonde, 2019-06-22:
Covered by #36

closed 2019-06-22 by SpenceKonde