SpenceKonde/megaTinyCore #34: Fill in feature table

SpenceKonde, 2019-06-19:
To avoid painting ourselves into a corner, we need a table of all the megaAVR attiny chips that are planned for support. This is a great way that people interested in megaAVR who aren't up to digging around in the core itself to help the project - hunt down datasheets, grab data, update spreadsheet, say something in this issue to let me know that an update has happened

This should list: 
Number of each timer available on each part. 
Amount of SRAM on each part
Presence/absence of DAC on each part (is this just all 1-series have a DAC, all 0-series do not? Is TCD like that too?)
Amount of EEPROM on each part

If you find any parameters that ought to be listed as well, add a column to the spreadsheet and add them.

Here is a google docs spreadsheet link. Editing is enabled (i'm trusting that everyone who bothers to see this will act in good faith and not vandalize it, hopefully I won't be disappointed in humanity)

https://docs.google.com/spreadsheets/d/1qALSjPBiLNx4_Er7RJg_b-INfTGehrhXASba9B7dShI/edit?usp=sharing

tedder, 2019-06-19:
I think this is a decent starting point to see the delta, it seems to be from before the 3216 existed:
http://ww1.microchip.com/downloads/en/AppNotes/Microchip%20AN2636%20Migration%20Between%20tinyAVR%201-Series%20and%20tinyAVR%200-Series%2000002636A.pdf

[image]

SpenceKonde, 2019-06-19:
Well that makes it easy...

SpenceKonde, 2019-06-19:
Filled in a bunch of stuff.

SpenceKonde, 2019-06-19:
Note that I can't figure out from the 214/414/814 datasheet whether the 414 has 64 or 128 bytes of EEPROM.

mraardvark, 2019-06-19:
I am sure some of this info can be harvested by a script from the ATDF files in the Microchip DFPs...?

mraardvark, 2019-06-19:
> Note that I can't figure out from the 214/414/814 datasheet whether the 414 has 64 or 128 bytes of EEPROM.

The .ATDF in ATtiny DFP 2.0.10 (2019-04-25) says:
<memory-segment exec="0" name="EEPROM" pagesize="0x20" rw="RW" size="0x0080" start="0x00001400" type="eeprom"/>

SpenceKonde, 2019-06-21:
Wonderful! 

How about someone whose time is not better used writing the core c code fill that table in? ;-)

closed 2019-06-22 by SpenceKonde