SpenceKonde/megaTinyCore #38: Get timer issue while testing PWM outputs

freemovers, 2019-06-20:
C:\Users\svandebor\Documents\Arduino\hardware\megaTinyCore\megaavr\cores\arduino\wiring.c:71:8: warning: missing terminating " character

 #error "No millis timer selected.

        ^

C:\Users\svandebor\Documents\Arduino\hardware\megaTinyCore\megaavr\cores\arduino\wiring.c:75:0: error: unterminated #if

 #if defined(MILLIS_USE_TIMERA0)

SpenceKonde, 2019-06-20:
Ugh. How did I not encounter this? (Oh, right, I only tested split tca
branch) Anyway, fix is simple. per is currently working on the travis ci,
when he finishes I'll fix this.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

freemovers, 2019-06-20:
Let me know if you had this fix in mind.

closed 2019-06-20 by freemovers

SpenceKonde, 2019-06-20:
Yeah, that's what was intended. Not testing the combined result after the merge was not my best work.

freemovers, 2019-06-20:
Unfortunately I am not getting the PWM outputs working at this moment. I also changed the boards.txt to DMILLIS_USE_TIMERA0, but I get other compiling issues as well, _timer is not declared  since TCA0.SPLIT.INTFLAGS is used instead?

SpenceKonde, 2019-06-20:
Let's not try to use timera for millis just yet. 

Are you able to get any code execution at all? I'm getting nothing!