SpenceKonde/megaTinyCore #39: Code does not execute, at all

SpenceKonde, 2019-06-20:
Something about what I did with TimerA seems to have hosed everything!

SpenceKonde, 2019-06-20:
WTF, I reverted those changes, but it still doesn't work!

SpenceKonde, 2019-06-20:
When was the last time this worked?!!

freemovers, 2019-06-20:
It was working fine using DMILLIS_USE_TIMERB0 instead of DMILLIS_USE_TIMERA0 in boards.txt

SpenceKonde, 2019-06-20:
Wait, I never even changed the millis timer and I can't get code to execute!

SpenceKonde, 2019-06-20:
No, it wasn't working fine then. I broke everything with 6567b71196ef8bd6b54397899dd0a34b127c880b

SpenceKonde, 2019-06-20:
You may have been getting build cache'ed

closed 2019-06-20 by SpenceKonde

SimonMerrett, 2019-07-15:
@SpenceKonde @freemovers Would this have something to do with why this won't compile for me:
```
#include <avr/io.h>
#include <avr/interrupt.h>

//------------------------------------------------
//TCA - 16 bit Timer/Counter Type A initialization
void timerInit(void)
{
  TCA0.SINGLE.INTCTRL = TCA_SINGLE_OVF_bm; // Enable timer interrupts on overflow on timer A
  TCA0.SINGLE.CTRLB = TCA_SINGLE_WGMODE_NORMAL_gc;
  TCA0.SINGLE.EVCTRL &= ~(TCA_SINGLE_CNTEI_bm);
  TCA0.SINGLE.PER = 0x0CB6;
  TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV256_gc
                      | TCA_SINGLE_ENABLE_bm;
}

//------------------------------------------------
ISR(TCA0_OVF_vect)
{
  // LED Toggling
  PORTA.OUTTGL = PIN6_bm; // toggle PIN 6, Port A
  // Clear the interrupt flag, to prevent instantly jumping back into the ISR again.
  TCA0.SINGLE.INTFLAGS = TCA_SINGLE_OVF_bm;
}

void setup() {
  PORTA.DIRSET = PIN6_bm;
  timerInit();
  sei(); // Enable global interrupts
}

void loop() {
}
```
and I get this:
```
Arduino: 1.8.9 (Windows 10), Board: "ATtiny412/402/212/202, ATtiny402, 20 MHz, 1.8V, Disabled, Disabled, EEPROM retained, 0.55V"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\_\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\_\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\_\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy2:chip=402,clock=20,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,dacvref=0v55 -vid-pid=1A86_7523 -ide-version=10809 -build-path C:\Users\_~1\AppData\Local\Temp\arduino_build_423453 -warnings=default -build-cache C:\Users\_~1\AppData\Local\Temp\arduino_cache_390010 -prefs=build.warn_data_percentage=75 -verbose C:\Users\_~1\AppData\Local\Temp\arduino_modified_sketch_951364\sketch_jul15a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\_\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\_\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\_\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy2:chip=402,clock=20,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,dacvref=0v55 -vid-pid=1A86_7523 -ide-version=10809 -build-path C:\Users\_~1\AppData\Local\Temp\arduino_build_423453 -warnings=default -build-cache C:\Users\_~1\AppData\Local\Temp\arduino_cache_390010 -prefs=build.warn_data_percentage=75 -verbose C:\Users\_~1\AppData\Local\Temp\arduino_modified_sketch_951364\sketch_jul15a.ino
Using board 'atxy2' from platform in folder: C:\Users\_\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.0.1
Using core 'arduino' from platform in folder: C:\Users\_\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.0.1
Detecting libraries used...
"C:\\Users\\_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny402 -DF_CPU=20000000L -DARDUINO=10809 -DARDUINO_attinyxy2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 "-IC:\\Users\\_\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\_\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\cores\\arduino" "-IC:\\Users\\_\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\variants\\txy2" "C:\\Users\\_~1\\AppData\\Local\\Temp\\arduino_build_423453\\sketch\\sketch_jul15a.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny402 -DF_CPU=20000000L -DARDUINO=10809 -DARDUINO_attinyxy2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 "-IC:\\Users\\_\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\_\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\cores\\arduino" "-IC:\\Users\\_\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\variants\\txy2" "C:\\Users\\_~1\\AppData\\Local\\Temp\\arduino_build_423453\\sketch\\sketch_jul15a.ino.cpp" -o "C:\\Users\\_~1\\AppData\\Local\\Temp\\arduino_build_423453\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\_~1\\AppData\\Local\\Temp\\arduino_build_423453\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
[... cut ...]
(.text+0x0): multiple definition of `__vector_8'

C:\Users\_~1\AppData\Local\Temp\arduino_build_423453\sketch\sketch_jul15a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ATtiny412/402/212/202.
```
I have tried with different clocks (20MHz and 5MHz). I took the code from the Microchip app note [Getting started with TCA](http://ww1.microchip.com/downloads/en/AppNotes/TB3217-Getting-Started-with-TCA-90003217A.pdf)

freemovers, 2019-07-15:
TCA is already used by Arduino for example for MILLIS(), so the interupt vector 8 (ISR(TCA0_OVF_vect)) is declared twice in your situation. I recommend using the RTC instead since that not used anywhere (see example in MD).

SpenceKonde, 2019-07-15:
as @freemovers said, TCA0 is used by the core - it provides all PWM (6 pins of it on chips with enough pins for it), as well as the millis()/micros() timing. 

Use one of the other timers if you need to take over a timer, especially if doing periodic interrupts. TCB0 (and TCB1 on boards that have it) would be a good choice IMO, as it's a 16-bit timer with lots of features, and not used for anything else. If you're willing to jump through the hoops associated with the async timer TCD0, that's also a good one.

SimonMerrett, 2019-07-15:
Thanks guys, I would like a fast timer so will try with TCB0.

commit 3a05eaa9 (2019-06-20): Move variant.c back, fix #39

variant.c must be in the variant folder. We need to figure out how to
move this stuff into another file and out of variant folder.