SpenceKonde/megaTinyCore #40: Figure out how to move variant.c to core

SpenceKonde, 2019-06-20:
There are no differences between the variants in this core, nor will there be, with regards to what needs to be in this file. But if I simply move variant.c into the core, it builds, but no code appears to execute.

SpenceKonde, 2019-06-20:
I think to figure this out I need to understand arduino builder better

kadamski, 2020-01-07:
@SpenceKonde  except the txy6 version seems to be missing isDoubleBondedActive(). Is  this intentional ?

SpenceKonde, 2020-01-07:
I noticed that, though it didnt seem to be throwing errors (I think there
is a second 'weak' definition of it that was getting used). Next batch of
changes coming fixed that and got rid of  I also managed to effectively
revert your wire fix in 1.1.3. Fixes on both of these are going into 1.1.5
(as well as changes around millis()/micros() that save another bunch of
space), which is coming out real soon now. I seriously need to get the
automated testing set up for megaTinyCore.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

commit 013b6213 (2020-01-09): Eliminate variant.c (fix #40 )

Remove isDoubleBondedActive, which was always false anyway, move
setup_timers() to wiring.c and eliminate variant.c for all variants
closed 2020-01-09 by SpenceKonde