SpenceKonde/megaTinyCore #42: Add An constants back in

SpenceKonde, 2019-06-20:
On reflection, I think removing these will create more problems than it solves. 

It's much simpler here though, since the ADC on the megaAVRs doesn't have differential channels (interesting - they dropped a feature that the classicAVRs had from the megaAVRs - I suspect the reasoning being that hardly anyone used them)

freemovers, 2019-06-20:
How should we number the A#, have it match the pin# as follows?
[image]

SpenceKonde, 2019-06-20:
An constants should resolve to the pin with ADC channel n on it, I think - the awkward bit is that AIN0 is on PA0 - the UPDI pin, which isn't available for use.

I am inclined to not put it on the pinout chart, as a way of discouraging people from doing it and encourage using the digital pin numbers.

freemovers, 2019-06-20:
I will take the analog of the pin chart, so the picture below is for reference to discuss #24, #6 and #44  

[image]

When this works I can start creating the pin-outs for the other sizes as well.

commit 5f162175 (2019-06-20): An style pin notation, Timer D for PWM

Fix #42. timer d listings in pins_arduino does *not* finish PWM on Timer
D.
closed 2019-06-20 by SpenceKonde