SpenceKonde/megaTinyCore #45: analogPinToBitMask and analogPinToBitPosition need to be fixed

SpenceKonde, 2019-06-20:
More mop-up of the moronic analog pin naming baggage imported from the official core. 

We got rid of the lookup table for analog channels. 

digitalPinToAnalogInput is probably all we need here? But someone needs to dig into how these are used by wiring_analog and hence what they need to be changed to to make them work.

SpenceKonde, 2019-06-21:
Maybe we can axe these macros entirely and adjust wiring_analog? 

But yeah, I think with the way it currently is, analogRead() is reading the wrong pin.

SpenceKonde, 2019-06-24:
Okay, this appears to work. I will have to look at the code again and figure out how on earth this could work. But work it does, so issue closed.

freemovers, 2019-06-24:
Using the pin# is not an issue, but using the A# is not working at this moment. For example analogRead(A10) throws an compiler error. Not a big deal in my opinion, but was this required to support to existing sketches?

SpenceKonde, 2019-06-24:
dafuk?! 

How can A0 not be visible from the sketch?! it's in pins_arduino.h. What do I need to do to make the A# defines visible from the sketch?!

SpenceKonde, 2019-06-24:
What makes even less sense is why this works in the official core when they also have A0 defined there!

freemovers, 2019-06-24:
My guess `const uint8_t PROGMEM analog_pin_to_channel[]` in pins_arduino.h is needed for An. No longer used for the regular pin#'s since you rewrote `#define digitalPinToAnalogInput(p)`?

SpenceKonde, 2019-06-24:
That would be entirely nonsensical if it solved it IMO - does it? Like, the problem is that all those static const uint8_t An = PIN_An constants aren't visible from the sketch... whether or not that array exists couldn't possibly do that, could it? 

I may defer this to 1.0.1 - I won't delay the release for it, but I have no idea how to fix it.

freemovers, 2019-06-24:
Moved the A# before the #ifdef ARDUINO_MAIN on the tx6, and now it compiles without errors.

SpenceKonde, 2019-06-24:
Good eye.

SpenceKonde, 2019-06-24:
My eyes completely passed over that #ifdef like it wasn't even there.

commit 42fcc93c (2019-06-24): Move An constants to the right place

That was dumb of me - I totally didn't realize that #ifdef was there.
Fix #45.
closed 2019-06-24 by SpenceKonde