SpenceKonde/megaTinyCore #53: Improve library examples

SpenceKonde, 2019-06-25:
Remove ones that aren't good examples and/or which don't compile on parts with small memory. 

If this leaves an absence of basic examples, create some.

per1234, 2019-06-25:
List of library examples that wouldn't compile for the 4x6 chips due to exceeding the available memory: https://github.com/SpenceKonde/megaTinyCore/issues/36#issuecomment-503854684

saysphilippe, 2020-01-21:
Hi Spence, It would be so awesome if you would create some SLEEP examples to show us how to do ex. STANDBY and PDOWN and react on an interrupt. I'm looking at using DS3231 or DS1337 for wake up towards the Attiny 1604 and1614. I have tried reading the chapter 11, but I'm not able to make that into actual code. :(

freemovers, 2020-01-21:
This might help to understand the use of sleep and interrupt:

```C
#include <avr/sleep.h>

#define PB4_INTERRUPT PORTB.INTFLAGS & PIN4_bm
#define PB4_CLEAR_INTERRUPT_FLAG PORTB.INTFLAGS &= PIN4_bm

ISR(PORTB_PORT_vect)
{
    if(PB4_INTERRUPT)                                           // PB4 changed state (pin#5 on ATtiny1616)
    {
        digitalWrite(2, CHANGE);                              // toggle the LED
        PB4_CLEAR_INTERRUPT_FLAG;
    }
}

void PORT_init(void)
{
  for (uint8_t pin = 0; pin < 8; pin++) {           // Disable the pull-up resistors to conserve energy
    (&PORTA.PIN0CTRL)[pin] = PORT_ISC_INPUT_DISABLE_gc; // Disable on PAx pin
#ifdef PORTB
    (&PORTB.PIN0CTRL)[pin] = PORT_ISC_INPUT_DISABLE_gc; // Disable on PBx pin
#endif
#ifdef PORTC
    (&PORTC.PIN0CTRL)[pin] = PORT_ISC_INPUT_DISABLE_gc; // Disable on PCx pin
#endif
  }
    
  PORTB.DIRCLR = PIN4_bm;
  PORTB.PIN4CTRL = PORT_PULLUPEN_bm;
  PORTB.PIN4CTRL |= PORT_ISC_BOTHEDGES_gc;
}

void setup()
{
  PORT_init();

  set_sleep_mode(SLEEP_MODE_PWR_DOWN);    // Set sleep mode to power down mode
  sleep_enable();                               // Enable sleep mode
  sei();                                         // Enable Interrupts

  pinMode(2, OUTPUT);                  // Set pin#2 as output 
}
 
void loop() {
    sleep_cpu();                               // Nothing to do here
}
```

SpenceKonde, 2020-01-21:
Worth noting that, once my rtc-millis option is ready (very soon), standby
sleep will be potentially very interesting, since the power consumption is
super low, but you can still use millis for timekeeping

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

saysphilippe, 2020-01-22:
Thank you for your example @freemovers! And Spence: We are waiting in suspense, hehe.

SpenceKonde, 2020-01-22:
Note that as of last night, the RTC millis function is in the github version, if that's what you're waiting in suspense for.

saysphilippe, 2020-01-22:
Whoho! Let's try it!

SpenceKonde, 2020-04-11:
2.1.0 will introduce some way cool stuff relating to sleep, including megaTinySleep which will let you use a civilized timer (ie, not the RTC) for millis, and switch to the RTC while sleeping so millis time is restored when it wakes back up. 

But that's not coming out for a while - I have another pressing matter to attend to between 2.0.0 and 2.1.0...

SpenceKonde, 2020-08-06:
Please open issues if other library issues are discovered. Please direct inquiries about sleep and millis-during-sleep to #158. Closing this issue as there are no known issues relating to this subject at this time.

closed 2020-08-06 by SpenceKonde