SpenceKonde/megaTinyCore #54: Reduce flash/memory use of core and libraries

SpenceKonde, 2019-06-25:
Current the core and it's libraries are flash hogs. This is problematic on the smaller ATtiny parts. We now create the assembler listing when dumping the compiled binary, so we have the tools we need to see where space is going and whether we can slim things down. 

It would be helpful to flag up any libraries or core functions that you find to be particular space hogs for special attention within this issue.

kadamski, 2019-12-19:
In my case (attiny416), just adding Wire.h as an include adds ~2300 bytes to the size of a program. If slave support is removed, this only adds ~1400 bytes so maybe it would make sense to either split master and slave implementation or make it possible to remove one?

Removing the possibility to change baud rate at runtime saves about 150 bytes. If we also constify the F_CPU_CORRECTED (which is currently always equal to F_CPU), another 250 bytes is saved.

kadamski, 2019-12-19:
BTW, we don't have to remove whole Slave functionality - we would just have to remove the Slave ISR (TWI0_TWIS_vect) as it always calls TWI_SlaveInterruptHandler() which brings some more Slave related funcitons as dependencies which cannot be removed from the binary. So an ifdef that enables/disables Slave/Master ISR could be useful.

SpenceKonde, 2020-01-17:
Serial can't get the same treatment without more careful investigation; based on discussion in #130 I am pretty sure this will break stuff, but that it could probably be bullied into working in most use cases. @kadamski were you able to make changes for Serial and compile sketches that used serial after doing so?

saysphilippe, 2020-02-15:
Hi! I'm trying to use the SoftwareSerial towards the Attiny202 but getting a compile error. When I deploy the same code to 1614 or 1604 there is no problem. I'm trying to use a 202 or 402 with a HC-12 to send a message. My application: Mailbox notifier. :)
The error message says:

c:/users/p/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avrxmega3/short-calls/crtattiny202.o:../../../../../crt1/gcrt1.S:311:(.init9+0x2): relocation truncated to fit: R_AVR_13_PCREL against symbol `exit' defined in .fini9 section in c:/users/p/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/avrxmega3/short-calls\libgcc.a(_exit.o)

collect2.exe: error: ld returned 1 exit status

Mange bibliotek ble funnet for "SoftwareSerial.h"
I bruk: C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.1.5\libraries\SoftwareSerial
exit status 1
Feil ved kompilering for kort ATtiny412/402/212/202.

SpenceKonde, 2020-03-19:
Somehow I missed this when it came through. Yeah - that's one form that running out of space can take (I think I even mention that in the readme). I think what's confusing the compiler is the unified address space.

I take it you've also looked into just using the builtin hardware serial port? In general I avoid software implementations of serial like the plague.

SpenceKonde, 2020-04-11:
Some improvements have been made here for 2.0.0. I don't see many more places to trim without impacting functionality. Closing this. Open specific issues if you spot bloat that I missed.

closed 2020-04-11 by SpenceKonde