SpenceKonde/megaTinyCore #57: Expanded docs to-do list

SpenceKonde, 2019-06-28:
We need a few general documents relating to megaAVR in general, since things people are used to doing on the classic AVRs won't work here. 

Specifically:
* Direct Port Manipulation (DirectPortManipulation.md) 
* Pin Change Interrupts (PCINT.md)
* Sleep/Power Save - is anything important different here? 
* ???

freemovers, 2019-07-08:
Added the Direct Port Manipulation and the Sleep/Power Save, with RTC example to wake-up from sleep.

SpenceKonde, 2019-07-08:
Thanks. I expanded the DirectPortManipulation one and linked them in from README.

SpenceKonde, 2019-09-03:
PinInterrupts is done, with an example.

closed 2019-09-03 by SpenceKonde

SpenceKonde, 2019-09-03:
Also just noticed and corrected error in PowerSave.md (INT32K is the internal ultra-low power oscillator, OSCULP32K, not  an external 32KHz crystal (XOSC32K)).