SpenceKonde/megaTinyCore #64: ATtiny1614

Taragorm, 2019-07-10:
`NUM_ANALOG_INPUTS` is #defined as 9 - this means you can't select A10/A11

freemovers, 2019-07-10:
The ATtiny 1614 has the following 10 analog channels on ADC0:
AIN0
AIN1
AIN2
AIN3
AIN4
AIN5
AIN6
AIN7
AIN10
AIN11
The UPDI programming pin is sharing AIN0, and cannot be used without disabling UPDI. Since that is not recommended we didn't assign the analog pin to UPDI, leaving you with 9 remaining analog input pins, A1 up to A7, A10 and A11. I did notice a typo for A11, and will get that corrected.

SpenceKonde, 2019-07-10:
As @freemovers said, that is correct - unless you're saying that analogRead(A10) doesn't work? 

A11 was non-working until @freemovers fixed it just now (this change is in github version but not released version)

Taragorm, 2019-07-12:
Yep - A10 doesn't work. As I noted, I had to increase `NUM_ANALOG_INPUTS`.

freemovers, 2019-07-12:
@Taragorm where did you notice a pin error, could you highlight below:
```
//                          _____ 
//                  VDD   1|*    |20  GND
// (nSS)  (AIN4) PA4  0~  2|     |19  10~ PA3 (AIN3)(SCK)(EXTCLK)
//        (AIN5) PA5  1~  3|     |18  9   PA2 (AIN2)(MISO)
// (DAC)  (AIN6) PA6  2   4|     |17  8   PA1 (AIN1)(MOSI)
//        (AIN7) PA7  3   5|     |16  11  PA0 (nRESET/UPDI)
// (RXD) (TOSC1) PB3  4   6|     |15  7~  PB0 (AIN11)(SCL)
// (TXD) (TOSC2) PB2  5~  7|_____|14  6~  PB1 (AIN10)(SDA)
```

freemovers, 2019-07-12:
I do not have an ATtiny1614 or 1604 to test this on, but I am sure that the 1616 most likely has the same issue, and I can verify that this weekend. For now you should be able to use the pin# instead:
`analogRead(6) // Reads A10`

freemovers, 2019-07-12:
@SpenceKonde since digitalPinToAnalogInput(p) already returns NOT_A_PIN when a pin# without analog channel is used, could we just eliminate the NUM_ANALOG_INPUTS check and just check if we have a channel# or NOT_A_PIN. We should probably rename pin to channel as well since it is very confusing.

SpenceKonde, 2019-07-12:
This should be fixed now. I thought I'd looked for cases where NUM_ANALOG_PINS was used, because that has caused problems for me on ATTinyCore, but somehow I missed this.

closed 2019-07-12 by SpenceKonde

Taragorm, 2019-07-16:
> @Taragorm where did you notice a pin error, could you highlight below:
>
> ```
> //                          _____
> //                  VDD   1|*    |20  GND
> // (nSS)  (AIN4) PA4  0~  2|     |19  10~ PA3 [...]

The pins on the RHS are labelled 14-20 - should these not be 8-14, on a 14 pin device?

Taragorm, 2019-07-16:
It looks like a 20-pin device has just been truncated...

Apologies for not being clearer!

freemovers, 2019-07-16:
Now I see, I was only focusing on the Arduino pin#'s. Thank you for getting back on this, it is corrected now.

commit 22d4e582 (2019-07-12): Fix #64 - correct valid pin check in analogRead

We do analog pins a little differently from the official core, so this test needed to be adjusted. I swear I searched the core for NUM_ANALOG_PINS and didn't find it used anywhere.