SpenceKonde/megaTinyCore #66: Cannot compile Blink.ino for ATtiny402

SimonMerrett, 2019-07-15:
Hi, I realise this is a new core but I thought I'd try it out on Arduino IDE 1.8.9 with my bare ATtiny402. Before I got to connecting any hardware, I checked compilation and got the following errors:
```
Arduino: 1.8.9 (Windows 10), Board: "ATtiny412/402/212/202, ATtiny402, 20 MHz, 1.8V, Disabled, Disabled, EEPROM retained, 0.55V"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\-1\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\-1\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\-1\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy2:chip=402,clock=20,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,dacvref=0v55 -ide-version=10809 -build-path C:\Users\-1\AppData\Local\Temp\arduino_build_757795 -warnings=default -build-cache C:\Users\-1\AppData\Local\Temp\arduino_cache_833 -prefs=build.warn_data_percentage=75 -verbose C:\Users\-1\AppData\Local\Temp\arduino_modified_sketch_959698\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\-1\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\-1\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\-1\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy2:chip=402,clock=20,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,dacvref=0v55 -ide-version=10809 -build-path C:\Users\-1\AppData\Local\Temp\arduino_build_757795 -warnings=default -build-cache C:\Users\-1\AppData\Local\Temp\arduino_cache_833 -prefs=build.warn_data_percentage=75 -verbose C:\Users\-1\AppData\Local\Temp\arduino_modified_sketch_959698\Blink.ino
Using board 'atxy2' from platform in folder: C:\Users\-1\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.0.1
Using core 'arduino' from platform in folder: C:\Users\-1\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.0.1
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny402 -DF_CPU=20000000L -DARDUINO=10809 -DARDUINO_attinyxy2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 "-IC:\\Users\\-1\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\-1\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\cores\\arduino" "-IC:\\Users\\-1\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\variants\\txy2" "C:\\Users\\-1\\AppData\\Local\\Temp\\arduino_build_757795\\sketch\\Blink.ino.cpp" -o nul
Error while detecting libraries included by C:\Users\-1\AppData\Local\Temp\arduino_build_757795\sketch\Blink.ino.cpp
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny402 -DF_CPU=20000000L -DARDUINO=10809 -DARDUINO_attinyxy2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 "-IC:\\Users\\-1\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\-1\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\cores\\arduino" "-IC:\\Users\\-1\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.1\\variants\\txy2" "C:\\Users\\-1\\AppData\\Local\\Temp\\arduino_build_757795\\sketch\\Blink.ino.cpp" -o "C:\\Users\\-1\\AppData\\Local\\Temp\\arduino_build_757795\\preproc\\ctags_target_for_gcc_minus_e.cpp"
avr-g++: error: device-specs/specs-attiny402: No such file or directory

exit status 1
Error compiling for board ATtiny412/402/212/202.
```
I'm happy to be a tester on the 402 but would very much appreciate suggestions on where to start looking for the cause of this error(s). Equally, if I there's a better way to give the information needed to help with this, please let me know.

freemovers, 2019-07-15:
Looks like you are missing the device-specs which are automatically added to your system when you install the Arduino megaAVR with the board manager (Uno Wifi Rev2).

SimonMerrett, 2019-07-15:
> Looks like you are missing the device-specs which are automatically added to your system when you install the Arduino megaAVR with the board manager (Uno Wifi Rev2).

You don't mean the one that the docs for this repo explicitly tell you install?! Doh! thank you. I'll check and close the issue if that works.

SimonMerrett, 2019-07-15:
@freemovers thanks again - adding the megaAVR boards worked. Found another issue specifically for the 402 signature but I'll put that separately. It's a Microchip hardware and avrdude problem so not sure if anyone wants to get involved for this repo...

closed 2019-07-15 by SimonMerrett

SpenceKonde, 2019-07-15:
I added a more prominent note about this on the readme.

lokeb, 2023-06-28:
Hi. I am getting the same error even after installing Arduino MegaAVR.

SpenceKonde, 2023-06-28:
Warning: platform.txt from core 'megaTinyCore' contains deprecated compiler.path={runtime.tools.avr-gcc.path}/bin/, automatically converted to compiler.path=/usr/bin/. Consider upgrading this core.
Detecting libraries used...
"/usr/bin/avr-g++"

You using a version of Arduino that was installed by a package manager? there's one package manager that tampers with the rewrite file, causing third party cores which specify a new toolchain version to be forced onto the old one (which doesn't work) instead of the new copy board manager installed for you - like they specifically broke board manager because they don't like what board manger does, even though that's necessary to make things work in the real world. 

Anyway, in the future please do not post new issues to long closed issuies, create a new discussion topic. when the thread was created, the bogus versions distributed with package managers were failing in very different ways. Never use Arduino installed from a linux package manager.