SpenceKonde/megaTinyCore #67: attiny402 device signatures. Do we want to workaround?

SimonMerrett, 2019-07-15:
This is a specific issue to the Attiny402 hardware and avrdude. Since avrdude doesn't have a repo that I'm aware we can contribute to, do we want to include something in the docs for these cores?

See here for [related issue](https://github.com/ElTangas/jtag2updi/issues/12) in the jtag2updi repo.

Bottom line is that avrdude.conf has the 402 device signature ending in 0x23, when in fact it is stated in the datasheet as 0x27 and to add complications, some devices that are definitely 402s have the device signature ending in 0x25 (I have these). Using the jtag2updi programmer with the megaTinyCore on 402s won't work until the device entry in avrdude.conf is edited to suit the actual signature of the chips you have.

@SpenceKonde @freemovers and others, please feel free to close this if you (understandably) see this as outside the realms of this repo. I'd be happy to write a .md doc to explain to users though.

SpenceKonde, 2019-07-15:
Ew ew ew! 

This is an easy fix since we're only using programmer def's supplied with this core though, just grab a corrected avrdude.conf and put it into the core and point platform.txt at it. This is the same thing ATTinyCore does.

SimonMerrett, 2019-07-15:
We have a choice - we can correct the avrdude.conf to to the datasheet value and leave it at that. But it doesn't help those with wrong signatures in silicon. So we _could_ add a 402w part (w = workaround) in addition to the main, datasheet-correct 402 in avrdude.conf. What would your preference be? I can make the avrdude.conf either way. Or some other way would be fine, if you let me know what to do!

SimonMerrett, 2019-07-15:
[avrdude_conf.zip](https://github.com/SpenceKonde/megaTinyCore/files/3393312/avrdude_conf.zip)

SpenceKonde, 2019-07-15:
Oh for fucks sake, you're saying that there are multiple versions of the silicon that report different signatures? 

Uuuuugh!

In that case, add a new part definition to avrdude.conf with the different signature(s) found in the wild.

Document the names of those, and what signature they correspond to, in your PR (or along with a file added to this issue). 
Is there anything we can link to with background, ideally a document from microchip where they admit to the cockup? 

Then, I'll add an extra option to the Tools -> Chip menu for each signature variant, and add a note to the readme.

SimonMerrett, 2019-07-15:
The attiny402w is defined with signature in the attached zip file of avrdude.conf. I will dig out some links to the provenance.

SimonMerrett, 2019-07-15:
[image]
This is from the Atmel Studio device pack download area. Needless to say, avrdude isn't quite up to speed yet. This only covers the 0x23 signature ending (which should be 0x27 according to datasheet). 

This is the [topic on avrfreaks](https://www.avrfreaks.net/forum/attiny-202402-was-simulator-issue-now-bad-device-id) where someone else had the same problem of the signature for clearly 402s returning as ending in 0x25. I also have 0x25s. The OP there raised an issue with Microchip but I don't think you can track issues publicly. Sadly, there's nothing mentioned in [silicon errata](http://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny202_402-Errata-and-Clarification-DS40002123A.pdf) about this.

SimonMerrett, 2019-07-16:
@SpenceKonde I can put the new version of avrdude.conf in a PR if that's easier than using the zip I attached above. Would you like it in megaTinyCore/megaavr/extras/ ?

SpenceKonde, 2019-07-19:
Why does the 402w definition there have so much more stuff than the 402 definition? 

`	
    memory "NVM"
        size      = 0x08;
        offset    = 0x1285; 
		# 1280 original guess for fuse base from jtag2updi repo (jtag2updi/source/NVM.h)
		# 1100 original guess for signature base from jtag2updi repo (jtag2updi/source/NVM.h)
		# 1180 first yielded signature bytes 1E9225
		# 11C0 yielded non-repeated signature bytes
		# LOCKBIT offset    = 0x128A; #appears to be lockbit (0xC5 = unlocked)
		# SYSCFG0 offset 	= 0x1285; #was consistent with default/reset fuse settings (1s in unused bit positions)
        page_size = 0x01;
        readsize  = 0x01;
    ;
	
	memory "CLKCTRL"
        size      = 0x20;
        offset    = 0x0060;
        page_size = 0x01;
        readsize  = 0x01;	
	;	
	
	memory "MCLKCTRLB"
        size      = 0x01;
        offset    = 0x0061;
        page_size = 0x01;
        readsize  = 0x01;	
	;
	
	memory "CPU"
        size      = 0x10;
        offset    = 0x0030;
        page_size = 0x01;
        readsize  = 0x01;	
	;
	
	memory "CPU.CCP"
        size      = 0x01;
        offset    = 0x0034;
        page_size = 0x01;
        readsize  = 0x01;	
	;
	
	memory "FUSE.SYSCFG1"
	    size      = 0x01;
        offset    = 0x1286;
        page_size = 0x01;
        readsize  = 0x01;	
	;
	
	memory "FUSES"
	    size      = 0x10;
        offset    = 0x1200;
        page_size = 0x01;
        readsize  = 0x01;	
	;
	
	memory "ADC_RES"
	    size      = 0x10;
        offset    = 0x0610; 
        page_size = 0x01;
        readsize  = 0x01;	
	;
	
	memory "ADC_RES"
	    size      = 0x10;
        offset    = 0x0610; 
        page_size = 0x01;
        readsize  = 0x01;	
	;`

It also sounds like we need one for sigs ending in 0x23?

SimonMerrett, 2019-07-19:
Erm, because I left in all the guff from when I was trying to do things with the 402 that I now know are done in better ways! My apologies. Maybe I should fork and make the pr, like I did with the jtag2updi repo. Would be neater.

I don't know about the 0x23. I haven't heard of anyone claiming their part has this address. We could make a 402e for erroneous? I'm inclined to omit it.

SpenceKonde, 2019-07-20:
Reopened, because this needs to be set in boards.txt as well. But build.mcu is used for both compilation and uploading, and the compiler won't recognize tiny402w. 

Also, the 402's currently shipping have the correct signature per datasheet - the ones I got yesterday have the correct sig.

SpenceKonde, 2019-07-24:
Removed bug label (this isn't a bug, this is atmel's screwup that we're working around), and *actually* reopened.

commit edb18604 (2019-07-19): Use the core-supplied avrdude.conf (fix #67)

There, this should work now.
commit 38889778 (2019-07-24): Add workaround for 402 signature issue

Naturally when I reopened the issue, what had seemed intractable became
obvious. Fix #67.
closed 2019-07-24 by SpenceKonde