SpenceKonde/megaTinyCore #69: PlatformIO

JonSilver, 2019-07-27:
@SpenceKonde thank you so much for creating this core. Do you have any guidelines for using it within PlatformIO? Is this even possible at present? If not, do you have any plans to integrate it into PIO?

Looking forward to seeing your prototyping boards on Tindie.

SpenceKonde, 2019-07-27:
I am not familiar with PlatformIO, so I don't know what issues there might
be. What would need to be done to work with PlatformIO?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

JonSilver, 2019-07-27:
I've never done the work myself to integrate a core into PlatformIO. However I know @MCUdude has worked with the PlatformIO lead @ivankravets to integrate his [MiniCore](https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md)

SpenceKonde, 2019-07-29:
I'm happy to do this, and responded to that effect in the issue on platformio repo, but no idea what needs to be done, and haven't heard back over there.

JonSilver, 2019-07-29:
@SpenceKonde I guess they're all a bit busy, but thank you for your efforts to make this integration happen. I searched and found [this conversation on the PIO community](https://community.platformio.org/t/how-to-add-a-different-arduino-core/4698/7) which may shed some light on the process and what's involved. I see @MCUdude follows this repo so perhaps he'll have something helpful to add (though I'm reticent about volunteering people I don't know for time consuming activities). 🙂

ivankravets, 2019-08-01:
Hi all. Please file a feature request here https://github.com/platformio/platform-atmelavr/issues

JonSilver, 2019-08-02:
Thanks @ivankravets. @SpenceKonde will you do this as author and originator of the core, or shall I?

MCUdude, 2019-08-20:
I'm not 100% what @ivankravets did behind the scenes when MiniCore was added, but I created all relevant [board manifest files](https://github.com/platformio/platform-atmelavr/commit/afaa7a183a92459dbe0eb61554930ef1e448e185#diff-31c0a9510b83b6710a1d1e7434c63789). I also provided some [decent information](https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md) on how you can use and configure MiniCore through platformio.ini. Feel free to copy my markdown file. IMO it's a positive thing that the most popular 3rd party Arduino AVR codes shares the same PlatformIO integration style.

ivankravets, 2019-08-20:
I've just filed a feature request https://github.com/platformio/platform-atmelavr/issues/160

We will add support soon.

SpenceKonde, 2019-08-21:
Wonderful, thanks - let me know if there's any way I can be of assistance. 

BTW: In the next day or two, I'm putting in some significant changes for the 412/212/402/202, to reflect the latest version of the datasheet, which revealed that there are 4 PWM capable pins, not just 1 like the original version advertised; you should be sure to get these changes in.

MCUdude, 2019-08-21:
> BTW: In the next day or two, I'm putting in some significant changes for the 412/212/402/202, to reflect the latest version of the datasheet, which revealed that there are 4 PWM capable pins, not just 1 like the original version advertised; you should be sure to get these changes in.

What? All out of nowhere three PWM output were "added"?! Atmel/Microchip's datasheet never ceases to surprise me...

SpenceKonde, 2019-08-21:
They weren't added, they were previously omitted :-P

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-08-22:
Okay, the changes I referenced above are in the code (they will be in 1.0.5 release, which will happen as soon as pinout diagrams are available).

iosengineer, 2020-05-26:
What is the status of this very desirable feature addition to platformio? I can see support for some mega-avr in there now, but there are only two board definitions to choose from. I build my own from scratch, so I'm not sure how to proceed.

SpenceKonde, 2020-05-26:
https://github.com/platformio/platform-atmelavr/issues/83
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

iosengineer, 2020-05-26:
Thank you. I lost the thread. Even though I commented on it before. *rolls eyes*. I'm subscribed to it now.

SpenceKonde, 2021-02-04:
Closing because there is nothing to do on my end here, last I heard it should be in the next release of AVR for platformio

closed 2021-02-04 by SpenceKonde

ivankravets, 2021-02-04:
Thanks to everyone who helped with this issue!