SpenceKonde/megaTinyCore #70: Pinout diagram for xy4 pin numbering wrong

SpenceKonde, 2019-07-29:
Pin numbers are wrong here - pin numbers 4 and 5 are missing, and all the numbers after that are offset by 2 (as one would expect).

SpenceKonde, 2019-07-29:
This issue is only in the docs - it's not present in the code.

SpenceKonde, 2019-08-04:
Arduino pin numbering is wrong on the xy4 past pin 3. 
Physical pin numbering is wrong on the xy2 on right hand side.

Image dimensions for the x0z and x1z is different - if it's not difficult, could you export them at the same dimensions? I would like to be able to embed both in one page, or stack the images above eachother, or whathaveyou, and have them appear the same size. (this is most relevant for the xy2). 

Thanks!

SpenceKonde, 2019-08-04:
Stopgap pinout images have been put in place pending a proper fix - they don't look quite as nice (particularly the x04 one), since I modified them in MS paint, but I needed accurate pinout images for listings for products launching next week.

freemovers, 2019-08-04:
@Spence, the original is on a single page (X2 to X6), is that what your are looking for? I can e-mail you the original GIF.

SpenceKonde, 2019-08-05:
Hurray - thanks! These look good!

These ones had the same dimensions for the 1-series and 0-series, so I was able to stack the x02 and x12. For some reason, before they were scaled differently, so I couldn't easily do that.

closed 2019-08-05 by SpenceKonde