SpenceKonde/megaTinyCore #72: Minor documentation points

technoblogy, 2019-08-03:
Thank you for this great work! At last I can program the new ATtiny Series 0 and Series 1 chips from the Arduino IDE!

A few minor notes:

* On the ATtiny202/402 and ATtiny212/412 pinout diagrams pins 11 to 14 should be 5 to 9.

* Introduction: "and the 1-series parts have a DAC for analog output as well". The ATtiny1614, ATtiny1616, ATtiny1617, ATtiny3216, and ATtiny3217 have 3 DACs.

* Introduction: "These parts all have 11 ADC channels available (except for the 8-pin parts, which have 5)". According to your tables the ATtinyx02 and ATtinyx12 have 6, the ATtinyx04 and ATtinyx14 have 10, and the ATtinyx06, ATtinyx07, ATtinyx16 and ATtinyx17 have 12.

* Direct Port Manipulation: Thanks for providing this - it's really helpful.

SpenceKonde, 2019-08-03:
Responses in order:

You mean the x14 and x04? I've already made issue for the pinout diagram, but haven't heard back from @freemovers who created the diagrams - I don't have the files needed to fix this myself

They have 3 DACs, but only o e of them can output its voltage on a pin. The others can only be used internally.

On the ADC channels,11 channels are *available* because the 12th is on the UPDI pin and isn't available for use. Like the classic avrs that have one on reset. I will think about how to better describe this.

Glad to hear that

technoblogy, 2019-08-03:
> You mean the x14 and x04?

No, the pins on the actual ATtiny202/402 and ATtiny212/412 chip package.

SpenceKonde, 2019-08-03:
Oooh. Yeah. I see

SpenceKonde, 2019-08-04:
These have been fixed - thanks for reporting :-)

closed 2019-08-04 by SpenceKonde