SpenceKonde/megaTinyCore #77: Need new pinout diagram for xy2

SpenceKonde, 2019-08-21:
They recently updated the datasheets. Turns out the pin mapping table was incomplete! I thought it was crazy that there was only one PWM channel

We need following things added: 
PWM on PA1,2,3,7 (not just PA3). 
UART pins on PA2,3 marked as alternate location, UART TX on PA6, UART RX on PA7. There will be a menu to select alternate or standard location for UART, since there are compelling use cases that will require using both mappings to avoid limiting the functionality of the part.

freemovers, 2019-08-21:
Does PWM work on timer TCD for these (WA0 on PA6)?

SpenceKonde, 2019-08-21:
It could be made to work, but I was planning not to implement that at this point, in light of the extremely tight flash on the xy2 series parts.

SpenceKonde, 2019-08-22:
I've confirmed that the new PWM pins and UART menu options now work. Will do a release as soon as I get the pinout diagrams.

SpenceKonde, 2019-08-23:
Wonderful! Thanks!

closed 2019-08-23 by SpenceKonde