SpenceKonde/megaTinyCore #81: Should we defang the F() macro?

SpenceKonde, 2019-09-03:
I was going to write a little blurb about PROGMEM on the megaavr's when I made a miraculous discovery. 

Because the flash in the megaavrs is memory mapped, any variable declared const is not copied to SRAM at initialization, and can be accessed and used like any other variable. 

If the variable is declared with PROGMEM keyword, however, the pointer will be offset from the bottom of flash, instead of the bottom of the whole address space - so normal access operations will fail, while pgm_read_type_near() will work. But there's generally no reason to use PROGMEM declarations anymore!

But that brings us to the subject of the F() macro. There is now NO BENEFIT IN RAM USE to using the F() macro: Serial.println("TEST") and Serial.println(F("TEST")) use the same amount of SRAM. However, what the F() macro *does* do is waste flash, because it declares the string PROGMEM, and then reinterprets cast to _flashStringHelper, which in turn uses pgm_read_byte_near() to read it. This of course leads to an obvious potential enhancement: 

F() is defined in api/String.h - can anyone (@per1234 ? ) think of any reason why it would not be an enhancement to change the definition of F() from
`#define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))`
to
`#define F(string_literal) (string_literal)`

In otherwords, a no-op for compatibility, without wasting space accessing the flash in what is now a roundabout and inferior way?

JonSilver, 2019-09-03:
Sounds like an excellent design decision, introducing a family-specific optimisation whilst maintaining code compatibility. This would have bitten me on the bum all over my existing code.

MCUdude, 2019-09-16:
This is[ already merged](https://github.com/arduino/ArduinoCore-megaavr/commit/d316aa62ae4f515095319cd9e8b0ac781daf5fb8)  in ArduinoCore-megaavr.

closed 2019-09-16 by SpenceKonde