SpenceKonde/megaTinyCore #83: Analog pin numbering on pinout diagrams clarification

hiduino, 2019-09-10:
It is not clear of the analog pin names on the pinout diagrams for the various chips.
For example on the xy7 diagram the analog pin label, digital pin '3' is not the same as saying 'A3'.  This is mapped to analog pin 'A7'.  Can this be clarified?  It's a bit confusing.

SpenceKonde, 2019-09-11:
Use of "A#" style pin names is strongly deprecated - having a second set of names for each pin just leads to more confusion; from the documentation:
> While the An constants (ex, A0) are supported, and refer to the corresponding ADC channel (not the corresponding pin number), using these is deprecated - the recommended practice is to pass the digital pin number to analogRead()

It was a _conscious decision_ to omit the An constants from the diagram to encourage use of the digital pin numbers. Is there a way that this could be made clearer? I would welcome input on this matter.

hiduino, 2019-09-11:
Hmm, okay good to know.  I was not aware of the depreciation effort.

hiduino, 2019-09-12:
Just curious, Arduino is still embracing the An designation as it is label on all their boards, so this will probably be still around in use, at least with Arduino boards.

closed 2019-09-11 by hiduino