SpenceKonde/megaTinyCore #86: chip not working

vicmaaa, 2019-09-13:
first, very appreciate your contribution of this core, but recently i use this core to successfully download a program into my board(i layout a board using attiny3217), however the mcu doesn't working at all, the current changes when i choose different clock frequency in arduino, still the led doesn't blink. i'm really out of thoughts, can you give me some advice? really appreciate.

SpenceKonde, 2019-09-13:
My attiny3217 boards work fine with blink (indeed, they come preprogrammed with it); you can see a copy of my schematic here for reference

[image]

Since the current changes when you change the clock frequency, that implies that you are indeed programming it successfully. That narrows the problem to either your code, or a wiring/assembly problem effecting the output pin but not the programming pin. 
Since you have not supplied your code or schematic, I can't point out any problems there

I think the most likely causes are: That you are using the incorrect digital pin number (see https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/ATtiny_x17.md ) for the pin you have the LED connected to - possibly you are using physical pin number instead? 

That the chip is not soldered on properly - those QFN parts are very hard to solder successfully without professional production equipment - I have had to rework about a third of the ones I have assembled at home (using stainless steel stencils and reflow oven - while i haven't had to do this on an SOIC part in over a year). Examine the pins under a magnifying glass for dry solder joints and bridged pins and use no-clean gel flux, small tip soldering iron, and drag soldering techniques if any issues are found.

Hopefully this will get things sorted for you; feel free to include your code and schematics if you'd like me to give them a look over. 
-Spence

vicmaaa, 2019-09-14:
thank you so much for your reply. this morning i figured out the cause of the problem: when i download the hex file generated in arduino, the board just doesn't work, as descripted yesterday; so i create a test code in avrstudio, then i download the hex file generated by avrstudio using cmd(avrdude -CC:\Users\xxx\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.0.5/avrdude.conf -v -pattiny3217 -cjtag2updi -PCOM6 -e -Uflash:w:F:\Users\ProjectAVR\blink\blink\Debug/blink.hex:i), the board works. so i guess maybe the complilation or linking process in arduino doesn't work fine(eg: some .h file has not been called correctly). is there any possibility that i have not installed the megatiny core correctly? but i install it throuth board manager. I believe you must have been generated a hex files in arduino that works fine on board, but still i can't figure out how to solve the problem. hope make myself clear. thanks again for your concern!

vicmaaa, 2019-09-14:
[image]
[image]

SpenceKonde, 2019-09-14:
Pin 21 (per the pinout diagram) is PA0/updi/rst pin (which can only be used
if fuses are set to make that in act as gpio, thus requiring high voltage
programmer to upload new code). This core does not support setting that
fuse, as most users would be unable to reprogram it afterwards.

I think you are using physical pin numbers instead of the logical pin
numbers (your schematic shows no led on digital pin 21, but does show led
on physical pin 21). Refer to the included pinout diagrams for the pin
numbers to use with digitalWrite/etc.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

vicmaaa, 2019-09-14:
That's really a low level mistake. Sorry to take you this trouble. I will verify it as soon as possible. Thank you for your patience.

vicmaaa, 2019-09-15:
I've confirmed my previous question is due to the wrong pin number, i put the physical pin num into the function incorrectly. BUT!!!! I've met a far more terrible problem: i change the mcu frequency in arduino, and burn bootloader, the i can no more download!! the screencut is shown bellow, happend on two boards. any suggestion?
[image]

vicmaaa, 2019-09-17:
OK, i've solved the problem. If you just begin to use avr's mcu, the fuse can really be confusing, when i modified the frequency in arduino, i modified the BOD value to approximately 3.3V, which i didn't notice. so when i burn the bootloader to take my modification into effect, the last problem occured. since the power supply of my board is 3.3V, maybe even lower. so, it's quite clear that the BOD prevents mcu to interact with programmer correctly. the solution is i put a 5V power supply into my board, then reburn the bootloader(BOD value set to 1.8V) successfully. up till now, my board with 3.3V power supply works normally.
As a newer into avr mcu developing, i suggest you should firstly look deep into the datasheet of your chip, especially the fuse related contents, that's really important.

closed 2019-09-17 by vicmaaa

SpenceKonde, 2019-09-20:
Thanks for updating with the solution. Also, be aware that the BOD values are not exact - if you look at electrical characteristics section, it allows for quite a bit of variation in the BOD voltage.