SpenceKonde/megaTinyCore #89: analogWrite() not working on PA7 - ATtiny402

SimonMerrett, 2019-09-18:
I have scanned the closed issues and found some hints that this issue maybe related to but nothing that directly matches my experience. Changing `led` from 0 through to 4, in this code, gives me PWM changes in LED brightness on pins 1-4 but nothing between full on (3 seconds) or full off (1 second). The pinout diagram shows pin 0 / PA7 is set as PWM capable so am I missing something?

```
// Test PWM and digital outputs
int led = 4; // works for pin 2 (PA1), pin 3 (PA2) and pin 4 (PA3) but not pin 1 (PA7)

void setup() {
  pinMode(led, OUTPUT);   // initialize the digital pin as an output.
}

void loop() {
  analogWrite(led, 127);    // turn the LED half way on 
  delay(1000);              // wait for a second
  digitalWrite(led, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
  digitalWrite(led, HIGH);  // turn the LED on by making the voltage HIGH
  delay(1000);              // wait for a second
  analogWrite(led, 15);     // turn the LED nearly all the way off
  delay(1000);              // wait for a second
}
```

SpenceKonde, 2019-09-20:
This pin is weird, in that the relationship between port pin number and timer is different from the other 0-series and 1-series parts. I'll look into this; it should be easy to sort out.

closed 2019-09-22 by SpenceKonde

SpenceKonde, 2019-09-22:
Thanks for reporting it! It was a bug with turnOffPWM() used by digitalWrite() to turn off PWM. Because of the way the numbering works out for the timer channels in all cases except that pin on this part, the bit of the pin in the port can be used to determine the timer channel - in analogWrite() I have a specific correction for the xy2 parts, but I forgot to do the same thing in turnOffPWM() for the different mapping on the xy2.

currycooqoo, 2023-09-25:
I'm also experiencing a similar issue with ATtiny 412.
Arduino IDE v1.8.19 works correctly. However, with PlatformIO v3.3.1 on VSCode v1.82.2, PWM did not work on PA07.
Thank you.

main.cpp
```cpp
#include <Arduino.h>

// Test PWM and digital outputs
int led = 4; // works for pin 2 (PA1), pin 3 (PA2) and pin 4 (PA3) but not pin 1 (PA7)

void setup() {
  pinMode(led, OUTPUT);   // initialize the digital pin as an output.
}

void loop() {
  analogWrite(led, 127);    // turn the LED half way on 
  delay(1000);              // wait for a second
  digitalWrite(led, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
  digitalWrite(led, HIGH);  // turn the LED on by making the voltage HIGH
  delay(1000);              // wait for a second
  analogWrite(led, 15);     // turn the LED nearly all the way off
  delay(1000);              // wait for a second
}
```

platformio.ini
```ini
; PlatformIO template configuration file for megaTinyCore
; https://github.com/SpenceKonde/megaTinyCore/
;
;   Build options: build flags, source filter
;   Hardware options: oscillator type, BOD, EEPROM retain
;   Upload options: custom upload port, speed, and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options
; https://github.com/SpenceKonde/megaTinyCore/blob/master/PlatformIO.md
; https://docs.platformio.org/page/projectconf.html
; https://docs.platformio.org/en/latest/platforms/atmelmegaavr.html

[platformio]
; Default build target
default_envs = Upload_UPDI

; Parameters used for all environments
[env]
platform = atmelmegaavr
framework = arduino

; Chip in use
board = ATtiny412
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L
; Oscillator type (internal or external)
board_hardware.oscillator = internal

; Unflag build flags
build_unflags =
; Extra build flags
build_flags =

; Monitor port is auto detected. Override here
;monitor_port =
; Serial monitor baud rate
monitor_speed = 115200

; Run the following command to upload with this environment
; pio run -e Upload_UPDI -t upload
[env:Upload_UPDI]
; Upload protocol for UPDI upload
upload_protocol = xplainedmini_updi
upload_port = /dev/cu.usbmodem14144202
upload_flags =

; run the following command to set fuses
; pio run -t fuses -e set_fuses
[env:set_fuses]
; Upload protocol for used to set fuses
upload_protocol = ${env:Upload_UPDI.upload_protocol}
upload_flags =
; Hardware settings
board_hardware.bod = 2.7v
board_hardware.eesave = yes
board_hardware.updipin = updi
```

SpenceKonde, 2023-09-25:
@currycooqoo Your issue is completely different. Please don't respond to long-closed issues. Unless you're on one of the first releases with a fix, and you think the fix isn't working... if you have a problem, it's not the same! Otherwise it would be fixed, and you wouldn't be having it.

And you also have brilliant flashing sign pointing to your PIO config file, since it doesn't happen with the IDE but does happen on PIO. It's because you are neither specifying the PWM mux menu option, nor providing the same define that the default option provides - which in this case tells the core initialization to put WO0 onto PA7, so WO3 can use PA3.  

This was supposed to be handled by setting the default for the part, and then having the non-default options override it - and it was for all parts other than the 8-pin ones. and the 14-pin non-optiboot ones - but you don't notice it there because the default option is empty.

Note that the menu that caused your problem was introduced 3.5 years after this issue was created >.>

commit 0a7aadbc (2019-09-22): Fix pwm on pin 1 on xy2 - fix #89

turnOffPWM() called by digitalWrite() when a pin may have PWM to turn it
off, failed to account for the unusual mapping pwm channels onto pins on
the 8-pin parts.