SpenceKonde/megaTinyCore #90: tone not working on some parts

technoblogy, 2019-09-20:
I can get tone working successfully on PA3 (Arduino pin 16) and PA5 (Arduino pin 1) on the ATtiny3216.

According to the datasheet the ATtiny202/402 has an output from TCB0 on PA6, Arduino pin 0, but I can't get tone to work on that pin.

According to the datasheet, on the ATtiny212/412 there is no output from TCB0 so tone can't be used on these parts; is that correct?

SpenceKonde, 2019-09-20:
Tone should work on all pins. It never uses hardware output compare on the megaAVR parts; there is no advantage or detriment to using a pin that supports hardware output compare. 

I am surprised to hear that it is not working on some pins!

As an aside, you're using the inaccurate "preliminary" version of the datasheet. The real version is here. The pin mapping table was missing a bunch of entries in the preliminary version. http://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny212-412-DataSheet-DS40001911B.pdf

technoblogy, 2019-09-20:
OK, I'll explore further and report back.

technoblogy, 2019-09-20:
Sorry, this was my faulty testing; it does indeed work on all pins as you said.

closed 2019-09-20 by technoblogy

SpenceKonde, 2019-09-20:
Thanks for retesting, glad to hear it's working.