SpenceKonde/megaTinyCore #94: Peripheral pin swapping not implemented

MCUdude, 2019-09-30:
The new tiny0/tiny1 series supports UART, TWI and SPI peripheral pin swapping. IIRC all tiny0/tiny1s has this feature except the 8-pin ones, where TWI can't be swapped.

It would also be nice if alternative peripheral pins were clearly marked in the colorful pinout diagrams.

SpenceKonde, 2019-10-11:
Agree it would be nice to have on the diagrams (we currently have it for the 8-pin series UART as I have that in a menu since I ship boards with it connected to the alt pins, since when I designed it the datasheet didn't say there was another option). @freemovers 

I guess it makes sense to put that menu option on all boards, esp once I get my shit in a row w/regards to optiboot, since I'd want to have a bootloader for both UART options... Not sure there's any need to do that for TWI/SPI though? Will the I2C/SPI libraries care if the user has set the pin swap register?

MCUdude, 2019-10-11:
> I guess it makes sense to put that menu option on all boards

Menu option? Why don't use mine/egilkv's implementation and use `Serial.swap()`, `Serial1.swap()`, `Wire.swap()` and `SPI.swap()`?

It is different for the bootloaders though. This is how I've implemented it.
[image]

SpenceKonde, 2019-10-11:
Is the .swap() function part of the standard Wire/SPI/Serial API? I didn't think it was...

MCUdude, 2019-10-11:
> Is the .swap() function part of the standard Wire/SPI/Serial API? I didn't think it was...

Nope, don't think so, and I don't think that it'll ever be a part of the official API because it doesn't really match the pinout of the Uno Wifi and Nano Every. But it's still a nice feature to have available. 
And our cores aren't really aimed at official Arduino boards.

(I've also added `printf()` to the print class, even though it will never be added to the official API. It's just because I find it very useful)

MCUdude, 2019-10-11:
But Serial.swap() and Serial.pins(,,) are at least available on the ESP8266.

MCUdude, 2019-10-13:
If adding extra features to the Arduino API bothers you, you can always just create a library that handles the peripheral pin swapping that the user can include.

SpenceKonde, 2019-10-14:
UART will be in a menu because it impacts bootloader selection. 

Since the pin swapping commands for other peripherals is a single command, this line will simply be documented.

SpenceKonde, 2019-10-14:
The UART menus are in. New issue for the docs (#97)

closed 2019-10-14 by SpenceKonde