SpenceKonde/megaTinyCore #95: const array not uploaded

bitluni, 2019-10-09:
Hi,
I have used an array defined like this:
`
const signed char soundsSamples[] = {.....
`
it's about 10kb in size. However the compiler seems to be not uploading the constant data. The sketch size is still 2%. Not errors while compiling. Accessing the array seems to return values had been stored in the flash section before.

using same array with PREGMEM works as expected.

Device is ATTiny1604
jtag2udpi as programmer

Any ideas what's wrong? The instructions mentioned that const should work fine

SpenceKonde, 2019-10-09:
That's not good! 

Can you post a sketch that reproduces this issue?

bitluni, 2019-10-10:
const w/o PROGMEM but with using pgm_read_ does not change the behavior.
I could imagine the size being a problem since other const arrays seem to work.
try this sketch. it will be always 950bytes compiled no matter how long the array is:
 
`void setup() 
{
}

const char a[] = {0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 
0, 0, 0, -1, 0, 1, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, -1, 0, 0, 0, 0, 1, 0, 0, -1, 0, 0, 0, 0, 0, 0, 0, 0, 0, -1, -1, 0, 0, 0, 0, 0, 0, 0, 1, 
0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, -1, -1, 0, 0, 1, 0, -1, -1, -1, 0, 0, 1, 1, -2, -1, 2, 1, -2, 0, 0, -1, 0, 0, 0, 0, 2, 1, -1, 0, 1, 0, 1, -1, -1, 0, -1, -1, 1, 0, 1, 0, 0, 1, 2, -1, -1, 0, 1, -1, 0, 0, -2, 
};

void loop() 
{
  for(int i = 0; i < 256; i++)
    digitalWrite(0, (PinStatus)a[i]);
}`

if you add volatile it will make the binary the proper size but it also consumes ram which is bad.

MCUdude, 2019-10-10:
I'm seeing the same behavior when compiling for the Arduino UNO Wifi Rev2 and the Arduino Nano Every too. Both are ATmega4809 based. It may be a good idea to open an issue over at the official [Arduino megaAVR core](https://github.com/arduino/ArduinoCore-megaavr) so that the Arduino devs can work on this.

SpenceKonde, 2019-10-11:
Well that's unfortunate! 

I guess we should figure out under what conditions const variables are left out by the compiler; it sounds like this is a compiler problem? 

Or is the core sending the wrong commands to the compiler and leaving out a section?

SpenceKonde, 2019-10-13:
Woah, woah, woah! 

Check this out: 

First, I added a few 1's near the beginning of the array. 
Second, I set the pin OUTPUT - which you had forgotten, so the current out of it when high would have been miniscule (very dim), and if wired such that LOW was on, would never turn the LED on. 
Thirdly, I added a delay to the loop just so you can see what was happening. 

> void setup(){pinMode(LED_BUILTIN,OUTPUT);}
> const char a[] = {0, 0, 0, 0, 0, 1, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -1, 0, 1, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, -1, 0, 0, 0, 0, 1, 0, 0, -1, 0, 0, 0, 0, 0, 0, 0, 0, 0, -1, -1, 0, 0, 0, 0, 0, 0, 0, 1,0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, -1, -1, 0, 0, 1, 0, -1, -1, -1, 0, 0, 1, 1, -2, -1, 2, 1, -2, 0, 0, -1, 0, 0,0, 0, 2, 1, -1, 0, 1, 0, 1, -1, -1, 0, -1, -1, 1, 0, 1, 0, 0, 1, 2, -1, -1, 0, 1, -1, 0, 0, -2,};
> void loop()
> {
>   for (int i = 0; i < 256; i++){
>     digitalWrite(LED_BUILTIN, (PinStatus)a[i]);
>     delay(100);
>   }
> }

This code works. 

However, we still get bogus sketch sizes: 

When uploading, we see that 1604 bytes of flash are uploaded:
avrdude: verifying ...
avrdude: 1604 bytes of flash verified

When COMPILING:
Sketch uses 1348 bytes (8%) of program storage space. Maximum is 16384 bytes.

So the problem is that the compiler is not including the size of const variables in the reported size of the sketch. 

From the readme, we list what we suspect to be a compiler bug. 
> Sometimes a sketch which is too big to fit will, instead of generating a message saying that, result in the error: 'relocation truncated to fit: R_AVR_13_PCREL against symbol tablejump2'.

I wonder if this is occurring when the sketch runs outside the memory limit only when counting the array!

SpenceKonde, 2019-10-13:
MUAHAHAHAAHAAAA! 

> void setup()
> {
>   pinMode(LED_BUILTIN,OUTPUT);
> }
> //Create a const array too big to fit in ram - 2048 bytes on a chip with 2048 bytes of ram, of which some is used elsewhere. 
> const unsigned long a[] = {2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2,2,1,1,1,2,2,1,2,2,2,2,2,1,1,1,2};
> void loop()
> {
>   for (int i = 0; i < 512; i++){
>     if(a[i]==2){
>       digitalWrite(LED_BUILTIN,HIGH);
>     } else if (a[i]==1) {
>       digitalWrite(LED_BUILTIN,LOW);
>     } else {
>       digitalWrite(LED_BUILTIN,LOW);
>       while(1) {;} //hang forever. If we read any value that's not supposed to be in the array, indicating something awful happening, the blinking will stop and we'll know there's a problem!
>     }
>     delay(100);
>   }
> }

Sketch uses 1350 bytes (8%) of program storage space. Maximum is 16384 bytes.
Global variables use 22 bytes (1%) of dynamic memory, leaving 2026 bytes for local variables. Maximum is 2048 bytes.

avrdude: 3398 bytes of flash verified

This sketch runs, and the LED does not hang - which would not be possible if it were actually putting the array in RAM and not telling us. 

1350+2048 = 3398, so the const array is not being reported in the sketch size!

Now, I pushed this further by extending the array from 2048 bytes to 15360, which would make the sketch too big to fit.... It compiles fine, but avrdude recognizes that it's trying to write to addresses out of range, and refuses to do so. 

avrdude: reading input file "C:\Users\Pete\AppData\Local\Temp\arduino_build_487809/arraytest.ino.hex"
avrdude: ERROR: address 0x4002 out of range at line 1025 of C:\Users\Pete\AppData\Local\Temp\arduino_build_487809/arraytest.ino.hex
avrdude: read from file 'C:\Users\Pete\AppData\Local\Temp\arduino_build_487809/arraytest.ino.hex' failed

SpenceKonde, 2019-10-13:
MUAHAHAHAHAH!!!!!!

C:\\Users\\Pete\\AppData\\Local\\Temp\\arduino_build_487809/arraytest.ino.elf  :
section                     size      addr
.data                          4   8402944
.text                       1346         0
**.rodata                    15360     34114**
.bss                          18   8402948
.comment                      17         0
.note.gnu.avr.deviceinfo      64         0
.debug_aranges               232         0
.debug_info                15643         0
.debug_abbrev               7309         0
.debug_line                 2642         0
.debug_frame                 156         0
.debug_str                  5577         0
.debug_loc                  1380         0
.debug_ranges                120         0
Total                      49868

The const array goes into .rodata instead of .data

But in platform.txt we were doing:

recipe.size.regex=^(?:\.text|\.data|\.bootloader)\s+([0-9]+).*

Changing that to 

recipe.size.regex=^(?:\.text|\.data|\.rodata|\.bootloader)\s+([0-9]+).*

fixes the problem! 

Man. Sometimes I'm so good I scare myself!

SpenceKonde, 2019-10-13:
I just put in a PR with the official arduino megavr core to fix this there too.

SpenceKonde, 2019-10-13:
@MCUdude  - you have a megaavr core that needs this fix too right?

closed 2019-10-13 by SpenceKonde

MCUdude, 2019-10-13:
@SpenceKonde Yes, I do. Would you like to submit a PR?

bitluni, 2019-10-13:
so it's only an output issue?

SpenceKonde, 2019-10-13:
Yeah, the only thing that is (well, was) broken is that the ide was
reporting the size of the sketch wrong, but compiling and uploading it
correctly.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

bitluni, 2019-10-13:
Thanks dudes, you are awesome!

Okt. 2019, 18:28:

MCUdude, 2019-10-13:
> Thanks dudes, you are awesome!

Well, thank _you_ for providing such awesome Youtube content! I'm just a lucky subscriber that gets to thank you in person! 🎉