SpenceKonde/megaTinyCore #96: Onboard mEDBG programmer setting  does not work for ATTiny416 Xplained Nano

EclipseBoom, 2019-10-12:
Tried to upload blink sketch using this option.  Seems the programmer always tries to use jtag2updi at least that's what the console message says.  Does not seem to try and use the xplainedmini_updi protocol.

Programmer is identified as Onboard mEDBG Uno Wifi Rev2 in the USB port list

AVRStudio can upload the .hex file so I know the hardware is OK (Nothing attached to Xplained board).

SpenceKonde, 2019-10-12:
Use upload using programmer.

This issue with default programmer for normal uploads will be fixed in next
release.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

EclipseBoom, 2019-10-12:
Thanks for the quick reply:

Using "Upload Using Programmer" with onboard mEDBG (tinyMegaCore) gives the following error: 

avrdude: Can't find programmer id "xplainedmini_updi"

if I choose the Uno Wifi Rev2 onboard mEDBG option I get:

`Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
An error occurred while uploading the sketch
avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0

avrdude done.  Thank you.`

For now I can manually track down the .hex file and upload via AVRStudio programmer tool but it would be nice to use the Aduino IDE if possible.

SpenceKonde, 2019-10-12:
Hm, actually 1.0.6 should have the fix for default programmer. 

But it looks like the problem at hand is something different anyway. 

I will look into this issue.

SpenceKonde, 2019-10-12:
Fix is in - but the current github version of the core is probably not in a usable state due to a bunch of half-done and untested stuff related to optiboot support. 

You should be able to pull the avrdude.conf from current version of github repo and replace the version included in the core (either in C:\Users\Pete\AppData\Local\Arduino15\packages\megatinycore\hardware\megaavr\1.0.6 for board manager install, or C:\Users\Pete\Documents\Arduino\hardware\megaTinyCore\megaavr for manual install) until there's a working 1.1.0 release available.

EclipseBoom, 2019-10-14:
Still doesn't work.  Replaced the avrdude.conf file from GIThub in both the local core area and the global area. Get the same error "Wrong UPDI fragment" or whatever.  

Windows 7 x64
Connection via USB to ATTiny416 Xplained Nano

Hardware verified to still work with AtmelStudio 7.0.2389 (latest update).

Did notice one thing.  Sometimes the mEDBG LED doesn't come on when trying from the Arduino IDE.  Always comes on and stays on with AtmelStudio.

I'm starting to wonder about driver issues but AtmelStudio still works correctly.

kadamski, 2019-12-19:
@SpenceKonde With the latest avrdude.conf as it is in git, I was still unable to flash my ATTiny416 Xplained Nano (with default firmware of mEDBG as it was never updated). I was getting such an error:

`avrdude: Device signature = 0x000038
avrdude: Expected signature for ATtiny416 is 1E 92 21
         Double check chip, or use -F to override this check.`

In order to fix it, I had to eaither remove the "has_pdi" from the .avr8x part (leaving only has_updi), or add has_pdi = no to the .avr8x_tiny part. 

Looking at the jtag3_initialize() in jtag3.c in avrdude, even if programer is UPDI (flag PGM_FL_IS_UPDI is set), the flag AVRPART_HAS_PDI may change the program flow as it is always checked before AVRPART_HAS_UPDI and the 2nd one is always in else if block.

So for example, the PARM3_ARCH is set to PARM3_ARCH_XMEGA instead of PARM3_ARCH_UPDI and a whole different PARM3_DEVICEDESC is set.

To sum up, it seems you can't have has_pdi = yes if you want to flash using updi.

fifthclassdesigns, 2021-11-11:
@SpenceKonde unfortunately, this appears to still be an issue using Arduino 1.8.13 and core 2.4.3 when uploading to the Xplained 416 board with the default settings:

```avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/nick/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.4.2/avrdude.conf"
         User configuration file is "/Users/nick/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0

avrdude done.  Thank you.

An error occurred while uploading the sketch
```

SpenceKonde, 2022-01-24:
Okay, this issue has been open for ages. This issue should be raised with them, it looks to be an avrdude bug, and I hear AVRdude is getting some maintenance currently.

closed 2022-01-24 by SpenceKonde