SpenceKonde/megaTinyCore #98: Add listing for alternate UART/SPI/TWI pins in diagrams

SpenceKonde, 2019-10-14:
We need alternate UART, SPI, and TWI pins marked on the pinout diagrams. 

This is your department @freemovers. 

Alt UART pins are already done for the xy2 parts, but we need them for other parts too. 
Alt TWI pins are not available for the xy2 parts. 

Thanks! 
(#94)

MCUdude, 2019-10-14:
By looking at the pinout diagrams, it's difficult to determine what pins represent what analog pin number? While you're at it @freemovers, how about adding the analog pin number to the pins as well (A0, A1 etc.)?

SpenceKonde, 2019-10-14:
We recommend using the digital pin numbers, as opposed to having two
separate sets of pin numbers for analog and digital reads. I dont really
want analog channel numbers shown unless the diagrams have a note that
their use is deprecated in favor of just using the normal digital pin
numbers with analog read.
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-10-23:
@freemovers - will you have a chance to look at this any time soon, or should I defer the diagram updates to 1.1.1?

freemovers, 2019-10-23:
I can work on this tonight. BTW, I had my work at an art festival last weekend using the ATtiny1616 to run APA102 strips with FastLED and 2 servo's simultaneous. It was great and not possible without this core, thank you so much!

SpenceKonde, 2019-10-25:
Awesome! Glad to hear it worked well!

MCUdude, 2019-10-25:
How about the CCL pins? I have a library that’s compatible with all tiny0/1’s that you can include or point to. It’s a very useful peripheral. I believe I read somewhere that it could be used to drive adressable LEDs directly?

SpenceKonde, 2019-10-25:
That's not a bad idea - though I am concerned at this point that by the time we're done, the pinout images will have so many labels as to be unreadable - @freemovers how crowded are the pinout labels looking?

SpenceKonde, 2019-10-25:
And, @MCUdude - sweet. I will totally steal your CCL library!

MCUdude, 2019-10-25:
You can have a look at my pinout diagrams. I've added almost everything relevant, and yes, they are kind of detailed. But hey, advanced (and excellent) features require some extra labels.

MCUdude, 2019-10-25:
@SpenceKonde please do! I've also made a pretty sweet analog comparator library. It's not ported to the tiny0/tiny1's just yet, but It's in the works. The top of the line tiny1's actually have three comparators!

MCUdude, 2019-10-25:
What's awesome about the comparators in the new megaAVRs is that they can use their internal reference/DAC voltage instead of a signal connected to the negative AC input. This means in many cases the user can omit the use of an external voltage divider. I'd call that a killer feature!

freemovers, 2019-10-25:
Will look like this:
[image]
Let me know if that contains everything and I will update this for the other configurations. Added the alternative pin locations for the USART, TWI and SPI and added LOGIC (CCL).

SpenceKonde, 2019-10-26:
I would change the legend from "Logic" to "Logic (CCL)" to clarify for
people not using the library. I dont see anything missing there.

Also, gotta love how 0 IN0 is only on the one pin virtually guaranteed not
to be usable x_x and that's the only CCL with all three inputs available
(though can one route a different pin to that with the event system? That's
the kind of thing the event system is able to do - but not relevant to
pinout diagram)

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

MCUdude, 2019-10-26:
You can connect the output of logic block 1 to input 0 on logic block 0 internally. So no big loss there, just provide a clear example

SpenceKonde, 2019-10-28:
@MCUdude That too. That'll be covered by the new documentation for the Logic library. 

So yeah - it sounds like what @freemovers has presented looks good. Make it so!

MCUdude, 2019-10-28:
> MCUdude That too. That'll be covered by the new documentation for the Logic library.

Yes. All the examples are designed to work with megaAVR-0 chips. I'm sure some/most examples will need modification to work with tiny0/tiny1 chips (pin layout etc.) Just make sure you test all examples on actual hardware before you release 1.1.0 😉

SpenceKonde, 2019-11-10:
@freemovers any update on this? 
I want to get 1.1.0 released in the next few days; the guy I gave a sample of the optiboot boards to at a party last weekend is asking me when he'll be able to play with it :-P

SpenceKonde, 2019-11-13:
Wonderful! Looks great!

closed 2019-11-13 by SpenceKonde