SpenceKonde/megaTinyCore #99: PTC support?

SpenceKonde, 2019-10-14:
I received an offer via email from someone who is interested in writing a library to make the PTC work; I had thought this to be impractical due to Atmel's strange behavior around the PTC library. 

I do not yet know whether this individual will be able to implement this, but if they can and did so, they would be a hero!

MCUdude, 2019-10-14:
Wow, that would be incredible! It should be possible to make it support mega0's too I believe.

I don't really understand why Atmel/Microchip is so secret about their Qtouch technology. The fact that the PC software creates a binary blob you have no control of makes me not want to use the built-in touch technology, and rather use [ADC touch](https://github.com/martin2250/ADCTouch) or a dedicated touch IC instead for commercial applications.

nicolasdauy, 2019-10-22:
What do you think of Xukangmin [TinyCore](https://github.com/xukangmin/TinyCore) ? It seems he has managed to implement the Touch on non-ADC pins. 
However it is harder for me to work with this Core because it has no board option, everything must be coded.

SpenceKonde, 2019-10-23:
His approach is sort of brute force - he generates the part specific libraries via atmel studio and links to them (much easier since he only supports two parts). It gives me some ideas though.

chipperdoodles, 2020-11-10:
Was thinking about trying out touch in some projects and looking up some Arduino alternatives to qtouch. Maybe porting something like: https://github.com/arpruss/ADCTouchSensor could be doable work in the mean time?

SpenceKonde, 2021-02-08:
I'm going to close this. If anyone else wants to do something here, I would welcome it.... but I have neither time nor interest in this,  let's admit it. This project is setting off all my "rathole" alarms.

closed 2021-02-08 by SpenceKonde

ladyada, 2021-06-16:
i may add it - leaving this as a note for myself
https://github.com/xukangmin/TinyCore/tree/master/avr/libraries/Touch

SpenceKonde, 2021-06-17:
It really does raise some questions re: why they are so secretive about it.... 
I think either A) there is some incredibly clever trick in the hardware that couldn't be hidden if you told people enough to implement it themselves, (seems unlikely tha this sort of magic might STILL be a concern, unless each generation has it;s own version of it. B) There is some incredibly ugly hackjob that would make people lose faith in it if they knew you were doing it, and it couldn't be hidden if they had to implement it (seems unlikely, it's been around long enough that they should have been able to sort those things out by NOW) the design is sufficiently byzantine that they don't think their customers are competent to implement it, or D) unlike customer-facing peripherals, it gets radical redesiigns frequently so customers would have to rewrite so much code that they wouldnt't migrate to new parts because of the effort involved in porting the PTC stuff. I think A) is unlikely. AVR machine code is readable enoughthat someone who really wanted to know would just examine the code and figure irt out.

SpenceKonde, 2022-03-22:
Note also that the PTC seems to have been dropped from all parts released after the DA-series. making development here even less attractive. 

On the other hand, the route towards support for someone fluent in AVR assembly is not particularly obscure. Disassemble the precompiled object files, reverse engineer them, and don't publish tem under your real name in case it could fall afoul of the DMCA or other oppressvive anti-innovation regulations.