SpenceKonde/megaTinyCore #104: Problems with flashing Attiny1604 after bugfixing v1.6

saysphilippe, 2019-10-30:
Hi!
I have reinstalled the board after you fixed the CRC problem, but since then and now I'm not able to flash my devices. It just times out in Arduino IDE. The setup is by the book with a 10 uF between Reset and GND, and a 4.7K Resistor between D6 and UPDI(PA0). I have tried to uninstall the board and reinstall it several times with no luck getting it to work again. I even have hooked up an Atmega328 instead of the Nano, and uploaded the jtag2updi sketch on both of them. Can you test in your end to see if you get it to work as it should?

Thanks in advance!

Philippe

SpenceKonde, 2019-10-30:
Weird. I have been using it throughout the development process without
problems, and it works on 1.1.0-dev too.

Are you using a breakout board that already has the 4.7k resistor on it
(like the ones I sell) (ie, resulting in two resistors in series) - this
might break it.

saysphilippe, 2019-10-30:
Hi, I'm using loose 4.7K resistors on a breadboard for alle of my instances. I have a simple breakout board for the ATTINY1604 with nothing else than the mcu. I have also replaced the resistors, and measured them with a multimeter :)

SpenceKonde, 2019-10-30:
Have you tried going back to 1.0.5 and seeing if that fixes the problem?

If not, that implies a hardware problem.

In any event, I can't look at this until middle of next week as I'm running
an event thurs ~ sun (no access to phone/email) and then have a priority
commission to work on.

saysphilippe, 2019-10-30:
1.05 gives the same CRC error when trying to install it. :( 1.04 get's installed but same problem with uploading code there as well. 
[image]

SpenceKonde, 2019-10-30:
1.0.5 gives a CRC error? dafuq? Unlike 1.0.6, which was added to the json
file by a dunderhead (me), 1.0.5 was added by per, and 100% worked in the
past. Am I so incompetent that I bunged the CRC for 1.0.5 too? Did I edit
the CRC for 1.0.5 instead of 1.0.6 when I c/p'ed 1.0.5 and turned it into
1.0.6? That would explain why 1.0.5 is hosed, if so.

But in any event, that 1.0.4 doesn't work confirms that this is a hardware
issue.

saysphilippe, 2019-10-30:
The one I'm trying to use now has old "blinky" code on it and it's blinking. I uploaded code sucessfully before the incident last week. I have also tried with several Nano's, and with several Attiny1604 on breakout boards(with no extra resistors.).

SpenceKonde, 2019-11-07:
Hmm. What is the exact error it gives?

saysphilippe, 2019-11-07:
It gives just a timeout after a while. like this. 
[image]

SpenceKonde, 2019-11-08:
What OS is this?

SpenceKonde, 2019-11-08:
I have another report of this issue now.

saysphilippe, 2019-11-08:
I'm running Windows 10 Pro 
[image]

saysphilippe, 2019-11-08:
Hi Spence! I sent you an email earlier if your would like me to assist by providing you a batch of 1604 so you can test them specifically. Just give me a heads-up and I will get some shipped to your location.

SpenceKonde, 2019-11-08:
Hi - yeah, I saw that, thank you. On the one hand, I want to say yes, because free stuff. But on the other hand, I also [have some inventory](https://www.tindie.com/products/drazzy/attiny16141604-dev-board-arduino-compatible/) so it's not like I need them!

Also, 1.0.5 will be fixed again tonight. I'm a bloody idiot and put the 1.0.6 checksum in for 1.0.5 (leaving the 1.0.5 checksum for 1.0.6, since I copy/pasted the whole entry) when I initially released 1.0.6, then got the correct checksum for 1.0.6 and fixed it, but didn't realize 1.0.5 checksum had the 1.0.6 checksum instead.

saysphilippe, 2019-11-08:
No worries, man! I'm monitoring your posts as it was the CNN, just eager to start using my 1604. But IF your need stuff, just let me know. It's the least I can do to get good libraries, to make our life worth living! hehe

SpenceKonde, 2019-11-09:
This isn't reproducing for me :-(

Was wondering if it was a bad avrdude version getting pulled in with a board manager installation, but it's working without issue for me - I burned bootloader and uploaded with no problems just now.

SpenceKonde, 2019-11-09:
Can you get the output with verbose uploads enabled?

SpenceKonde, 2019-11-09:
(btw, crc issue in 1.0.5 is finally fixed, and the script that copies the latest version of the json file to my website every night should be running now (it wasn't in the crontab. Nothing was in the crontab.))

saysphilippe, 2019-11-10:
Hi, I installed the 1.0.5 and no luck, sorry. I even uploaded the jtag2updi code to the Nano first as well. I put in a LED on D6 to see if it actually tried to send code, and it light up. I removed the LED and and exchanged the 4.7K resistor and tried again. :(

SpenceKonde, 2019-11-14:
Can you enable verbose output and give me the output it prints when you try and fail to upload to it?

saysphilippe, 2019-11-14:
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\p\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\p\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\p\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy4:chip=1604,clock=20,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,dacvref=0v55,millis=enabled,neopixelport=porta -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\p\AppData\Local\Temp\arduino_build_83378 -warnings=all -build-cache C:\Users\p\AppData\Local\Temp\arduino_cache_516722 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\p\AppData\Local\Temp\arduino_modified_sketch_830926\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\p\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\p\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\p\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy4:chip=1604,clock=20,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,dacvref=0v55,millis=enabled,neopixelport=porta -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\p\AppData\Local\Temp\arduino_build_83378 -warnings=all -build-cache C:\Users\p\AppData\Local\Temp\arduino_cache_516722 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\p\AppData\Local\Temp\arduino_modified_sketch_830926\Blink.ino
Using board 'atxy4' from platform in folder: C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.0.6
Using core 'arduino' from platform in folder: C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.0.6
Detecting libraries used...
"C:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny1604 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_attinyxy4 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 -DNEOPIXELPORT=VPORTA.OUT "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.6\\cores\\arduino" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.6\\variants\\txy4" "C:\\Users\\p\\AppData\\Local\\Temp\\arduino_build_83378\\sketch\\Blink.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny1604 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_attinyxy4 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 -DNEOPIXELPORT=VPORTA.OUT "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.6\\cores\\arduino" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.0.6\\variants\\txy4" "C:\\Users\\p\\AppData\\Local\\Temp\\arduino_build_83378\\sketch\\Blink.ino.cpp" -o "C:\\Users\\p\\AppData\\Local\\Temp\\arduino_build_83378\\preproc\\ctags_t [... cut ...]
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
Det oppstod en feil under opplasting av skissen

saysphilippe, 2019-11-17:
Hi man!
I just downloaded the new 1.1.0 version and this is the output:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\p\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\p\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\p\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy4o:chip=1604,clock=20,dacvref=0v55,millis=enabled,neopixelport=porta,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,uartpins=standard,twipins=standard,resetpin=UPDI -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\p\AppData\Local\Temp\arduino_build_473365 -warnings=all -build-cache C:\Users\p\AppData\Local\Temp\arduino_cache_228433 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\p\AppData\Local\Temp\arduino_modified_sketch_297712\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\p\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\p\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\p\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy4o:chip=1604,clock=20,dacvref=0v55,millis=enabled,neopixelport=porta,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,uartpins=standard,twipins=standard,resetpin=UPDI -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\p\AppData\Local\Temp\arduino_build_473365 -warnings=all -build-cache C:\Users\p\AppData\Local\Temp\arduino_cache_228433 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\p\AppData\Local\Temp\arduino_modified_sketch_297712\Blink.ino
Using board 'atxy4o' from platform in folder: C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.1.1
Using core 'arduino' from platform in folder: C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.1.1
Detecting libraries used...
"C:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny1604 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_attinyxy4 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 -DNEOPIXELPORT=VPORTA.OUT "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.1\\cores\\arduino" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.1\\variants\\txy4" "C:\\Users\\p\\AppData\\Local\\Temp\\arduino_build_473365\\sketch\\Blink.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny1604 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_attinyxy4 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 -DNEOPIXELPORT=VPORTA.OUT "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.1\\cores\\arduino" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.1\\variants\\txy4" "C:\\User [... cut ...]
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x42
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x42
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x42
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x42

avrdude done.  Thank you.

Problemer ved opplasting til kortet. Se http://www.arduino.cc/en/Guide/Troubleshooting#upload for forslag.

saysphilippe, 2019-11-22:
New verbose after update:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\p\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\p\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\p\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy4:chip=1604,clock=20,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,dacvref=0v55,millis=enabled,neopixelport=porta,uartpins=standard,twipins=standard -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\p\AppData\Local\Temp\arduino_build_625875 -warnings=all -build-cache C:\Users\p\AppData\Local\Temp\arduino_cache_684093 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\p\AppData\Local\Temp\arduino_modified_sketch_465781\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\p\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\p\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\p\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy4:chip=1604,clock=20,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,dacvref=0v55,millis=enabled,neopixelport=porta,uartpins=standard,twipins=standard -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\p\AppData\Local\Temp\arduino_build_625875 -warnings=all -build-cache C:\Users\p\AppData\Local\Temp\arduino_cache_684093 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\p\AppData\Local\Temp\arduino_modified_sketch_465781\Blink.ino
Using board 'atxy4' from platform in folder: C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.1.2
Using core 'arduino' from platform in folder: C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.1.2
Detecting libraries used...
"C:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny1604 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_attinyxy4 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 -DNEOPIXELPORT=VPORTA.OUT "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.2\\cores\\arduino" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.2\\variants\\txy4" "C:\\Users\\p\\AppData\\Local\\Temp\\arduino_build_625875\\sketch\\Blink.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny1604 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_attinyxy4 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERA0 -DNO_EXTERNAL_I2C_PULLUP -DDACVREF=0 -DNEOPIXELPORT=VPORTA.OUT "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.2\\cores\\arduino" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\1.1.2\\variants\\txy4" "C:\\Users\\p\\AppData\\Local\\Temp\\arduino_build_625875\ [... cut ...]
         Vtarget         : 5.0 V

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: jtagmkII_reset(): timeout/error communicating with programmer (status -1)
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)

avrdude done.  Thank you.

SpenceKonde, 2019-11-22:
I'm not sure what to say, as I've been programming these parts with a
jtag2updi programmer without issue, and am still able to, with 1.0.6 and
1.1.2 (and 1.1.0/1.1.1 where the bugs in those versions don't interfere);
unless I'm mistaken I've done this in the past week with 1.1.2 installed
both via board manager and manually.

I will note that I've had at least one jtag2updi programmer spontaneously
fail, with no indication of what went wrong or why. I cannot explain why
that happened (the programmer in question was made on a pro mini, and
appears to be completely dead, precluding most investigation) - but it
fails with this symptom; unless I'm mistaken that's the error message you
get when it is getting no communication from the jtag2updi programmer (ie,
it's the same error you'd get if you selected a serial port with nothing
connected to it).

-Spence

saysphilippe, 2020-01-16:
Hey man! I have finally figured it out. It is something wrong with some of my Arduino Nano's (168). I have tried four different Nano's and upload the jtac2pdi and then try to upload code to either 1604 or 1614 and two Nano's will not work for some reason. I'm not able to understand why, but as a wise man told me many years ago: Not all things are meant to be understood. We can now close this issue. :) Thanks for all the help!

closed 2020-01-16 by saysphilippe

SpenceKonde, 2020-01-17:
It wasn't by any chance not installing the 10uF cap between reset and ground was it? Using a board with a bootloader can require the cap (whether it is needed may depend on the bootloader on the nano, as the direct cause is the time that it stays in the bootloader after reset from the serial adapter's DTR line via the autoreset circuit)