SpenceKonde/megaTinyCore #106: Logic Library examples

SpenceKonde, 2019-11-08:
The examples are currently designed for the 0-series atmega devices, need to make sure they work with 0-series and 1-series attiny devices.

freemovers, 2019-11-18:
While working on some examples for the LOGIC library I noticed the input sources on the ATtiny are somewhat different from the MEGA. The current inputs in the library are as follows:

```
    masked       = 0x00,
    unused       = 0x00,
    disable      = 0x00,
    feedback     = 0x01,
    link         = 0x02,
    event_a      = 0x03,
    event_b      = 0x04,
    input        = 0x05,
    ac           = 0x06,
    input_pullup = 0x07,
    usart        = 0x08,
    spi          = 0x09,
    tca0         = 0x0A,
    tcb          = 0x0C,

```
According to page# 452 of the datasheet (ATtiny1616) the input sources are as follows:
```

0x0     MASK                                Masked input
0x1     FEEDBACK                       Feedback input
0x2     LINK                                  Linked other LUT as input source
0x3     EVENT0                            Event input source 0
0x4     EVENT1                            Event input source 1
0x5     IO                                       I/O pin LUTn-IN1 input source
0x6     AC0                                   AC0 OUT input source
0x7     TCB0                                 TCB WO input source
0x8     TCA0                                 TCA WO1 input source
0x9     TCD0                                 TCD WOB input source
0xA     USART0                            USART TXD input source
0xB     SPI0                                   SPI MOSI input source
0xC     AC1                                   AC1 OUT input source
0xD     TCB1                                 TCB 1 W0 input source
0xE     AC2                                   AC2 OUT input source
```

SpenceKonde, 2019-11-18:
That looks like a bug - does your experience with the hardware confirm your
suspicion that this is wrong for the megatiny parts?

freemovers, 2019-11-18:
Yes, I noticed this since I was working on an example using two timers (TCB and TCD) to create an output. Since the megaAVR has no TCD, they use a different CCL LUT input MUX (page# 357 of the megaAVR datasheet). 
Not sure how this library works. All examples are using input I/O, and after I selected a timer instead it threw and error that the input pin does not exist. Might just create some examples with the I/O first before moving to the other input sources.

MCUdude, 2019-11-18:
For reference, here's the input select struct for ATtiy3217 (from iotn3217.h). Not sure if all these options are available on other ATtinys too.

I chose to use numbers rather than the definitions (below) in the library source just because of readability. From a portability standpoint or would probably be better to use the definitions below, and add functionality like TCD by using macros.

```c
/* LUT Input 0 Source Selection select */
typedef enum CCL_INSEL0_enum
{
    CCL_INSEL0_MASK_gc = (0x00<<0),  /* Masked input */
    CCL_INSEL0_FEEDBACK_gc = (0x01<<0),  /* Feedback input source */
    CCL_INSEL0_LINK_gc = (0x02<<0),  /* Linked LUT input source */
    CCL_INSEL0_EVENT0_gc = (0x03<<0),  /* Event input source 0 */
    CCL_INSEL0_EVENT1_gc = (0x04<<0),  /* Event input source 1 */
    CCL_INSEL0_IO_gc = (0x05<<0),  /* IO pin LUTn-IN0 input source */
    CCL_INSEL0_AC0_gc = (0x06<<0),  /* AC0 OUT input source */
    CCL_INSEL0_TCB0_gc = (0x07<<0),  /* TCB0 WO input source */
    CCL_INSEL0_TCA0_gc = (0x08<<0),  /* TCA0 WO0 input source */
    CCL_INSEL0_TCD0_gc = (0x09<<0),  /* TCD0 WOA input source */
    CCL_INSEL0_USART0_gc = (0x0A<<0),  /* USART0 XCK input source */
    CCL_INSEL0_SPI0_gc = (0x0B<<0),  /* SPI0 SCK source */
} CCL_INSEL0_t;

/* LUT Input 1 Source Selection select */
typedef enum CCL_INSEL1_enum
{
    CCL_INSEL1_MASK_gc = (0x00<<4),  /* Masked input */
    CCL_INSEL1_FEEDBACK_gc = (0x01<<4),  /* Feedback input source */
    CCL_INSEL1_LINK_gc = (0x02<<4),  /* Linked LUT input source */
    CCL_INSEL1_EVENT0_gc = (0x03<<4),  /* Event input source 0 */
    CCL_INSEL1_EVENT1_gc = (0x04<<4),  /* Event input source 1 */
    CCL_INSEL1_IO_gc = (0x05<<4),  /* IO pin LUTn-N1 input source */
    CCL_INSEL1_AC0_gc = (0x06<<4),  /* AC0 OUT input source */
    CCL_INSEL1_TCB0_gc = (0x07<<4),  /* TCB0 WO input source */
    CCL_INSEL1_TCA0_gc = (0x08<<4),  /* TCA0 WO1 input source */
    CCL_INSEL1_TCD0_gc = (0x09<<4),  /* TCD0 WOB input source */
    CCL_INSEL1_USART0_gc = (0x0A<<4),  /* USART0 TXD input source */
    CCL_INSEL1_SPI0_gc = (0x0B<<4),  /* SPI0 MOSI input source */
} CCL_INSEL1_t;

/* LUT Input 2 Source Selection select */
typedef enum CCL_INSEL2_enum
{
    CCL_INSEL2_MASK_gc = (0x00<<0),  /* Masked input */
    CCL_INSEL2_FEEDBACK_gc = (0x01<<0),  /* Feedback input source */
    CCL_INSEL2_LINK_gc = (0x02<<0),  /* Linked LUT input source */
    CCL_INSEL2_EVENT0_gc = (0x03<<0),  /* Event input source 0 */
    CCL_INSEL2_EVENT1_gc = (0x04<<0),  /* Event input source 1 */
    CCL_INSEL2_IO_gc = (0x05<<0),  /* IO pin LUTn-IN2 input source */
    CCL_INSEL2_AC0_gc = (0x06<<0),  /* AC0 OUT input source */
    CCL_INSEL2_TCB0_gc = (0x07<<0),  /* TCB0 WO input source */
    CCL_INSEL2_TCA0_gc = (0x08<<0),  /* TCA0 WO2 input source */
    CCL_INSEL2_TCD0_gc = (0x09<<0),  /* TCD0 WOA input source */
    CCL_INSEL2_SPI0_gc = (0x0B<<0),  /* SPI0 MISO source */
} CCL_INSEL2_t;

```

SpenceKonde, 2020-01-04:
deferring to 1.1.4 as I don't really have time to dig into this and want to get the fixes and changes in 1.1.3 out to our users

SpenceKonde, 2020-02-13:
too many more serious bugs have cropped up in 1.1.5 to work on a relatively low priority like this.

closed 2020-03-18 by SpenceKonde

MCUdude, 2020-03-24:
FYI I just read at the Avrfreaks forum that the UPDI pin can in fact be used as an input even if the UPDI interface is active. However, the programming logic is always "listening", so it should be fine as long as one doesn't send a UPDI sequence.

But this is perfect for the Logic library, since one of the inputs are connected to PA0! You might want to point this out somewhere. Here's where I read it: [https://www.avrfreaks.net/comment/2744691#comment-2744691](https://www.avrfreaks.net/comment/2744691#comment-2744691)

commit 619642c9 (2020-03-18): Fix #106

Correct functionality on ATtiny parts. Correct examples for ATtiny
parts. Demonstrate use of event system to overcome limits on IN pins on
ATtiny parts in examples.

I now know more than I ever needed to about the CCL system (and a few
more things about some other peripherals). I'm not sure how useful it is
for Arduino users, but it should now work!