SpenceKonde/megaTinyCore #109: Enhancement: possible to add USB serial to Arduino-based UPDI programmer?

EclipseBoom, 2019-11-14:
Wondering if this is possible and how hard it would be.  I'd be willing to work on it some if there is a possibility.  Perhaps it needs a Leonardo/Mini or other 32U4 chip so the USB and UART are not both using the UART pins.

USB -> Host for UPDI and serial
UPDI -> D6 to target
D0/D1 -> UART to target

SpenceKonde, 2019-11-15:
Definitely possible with a 32u4 or other chip with native USB - but considerable effort would be involved, possibly including changes to avrdude to support this new programmer. Said effort needs to come from someone who is not me, as I have way too many irons to fire right now. The repo involved is here: https://github.com/SpenceKonde/jtag2updi

Is this even still a priority now that we have Optiboot serial bootloader support?

closed 2019-11-15 by SpenceKonde

MCUdude, 2019-11-15:
The [microUPDI programmer](https://github.com/MCUdude/microUPDI) is pretty close to what you want?

EclipseBoom, 2019-11-18:
Perfecto!  Thanks!  I always say everything you want is out there - you just have to find it...  That is proven true many times.