SpenceKonde/megaTinyCore #110: Add menu items for alt pins for I2C and SPI ports

EclipseBoom, 2019-11-14:
Since you already have this for the UART how about adding menus for the I2C and SPI ports?

SpenceKonde, 2019-11-15:
Ugh, fine.

SpenceKonde, 2019-11-15:
Done.

closed 2019-11-15 by SpenceKonde

MCUdude, 2019-11-15:
I don't really understand why this functionality has to be in the Tools menu in the first place. Changing to an alternative pin position can be done in the application at run time, so a simple one-liner would be better in any way IMO.

Another down side with this: Let's pretend I have a project file that uses alternative i2c pin position. Instead of having this setting _in the code_ I'd now have it in the tools menu. This means I'd manually have to change the i2c pins menu option, since the IDE project files don't store any of the information from the Tools menu.

SpenceKonde, 2019-11-18:
There's no pleasing everyone! 

But I did realize something that speaks in favor of menu options; the core provides constants for the pin numbers for SDA/SCL, and the SPI pins. These cannot be made correct without a tools menu option, and may be relied upon by libraries to operate on the correct pins (for example to set output/input, and the like). 

I think that this solution

EclipseBoom, 2019-11-18:
I think the Teensy standard is to declare the pins in the begin method.  The pins selected can be checked at compile time to make sure they are valid Serial.begin(14, 15): or symbolic wire.begin(NORMAL_PINS); or wire.begin(ALT_PINS);

This would allow kind of like 2x I2C, UART or SPI busses to be multiplexed with one peripheral block.  Probably not super critical on these small CPUs.  Manipulating the PORTMUX directly is not hard it just isn't abstracted in the usual Arduino way.  

OTH these little chips don't have the luxury of lots of C++ abstraction and layers of classes above the hardware.

So on balance my two cents are to leave it in the menu.  I'm guessing the number of times people will use both set of pins or want to swap in software is very small.