SpenceKonde/megaTinyCore #114: Make bootloader option as a sub menu

MCUdude, 2019-11-16:
It should be a fairly straight forward job to add a bootloader menu option as a sub menu rather than having duplicates in the boards menu. Is there a good reason to keep it this way?

SpenceKonde, 2019-11-17:
My current "reason" for keeping it this way is that there is at least one menu option that is not used for optiboot configurations, namely the EESAVE option (which is not shown for optiboot board defs because when using optiboot, the EEPROM is not erased during reprogramming

MCUdude, 2019-11-17:
I have a similar quirk in the MegaCoreX tools menu. When "Bootloader" is selected the functionality of the reset pin (also an option in the tools menu) is overridden. I just added this "bug" to the [readme](https://github.com/MCUdude/MegaCoreX#using-a-bootloader).

SpenceKonde, 2019-11-18:
Oh, good point - the reset pin function when non-bootloader boards are in use would have to be ignored as well, since if it is set to reset, further programming (such as, uploading the sketch) is disabled.

MCUdude, 2019-11-18:
Yes, you could basically just have one menu to Bootloader + choose what function the UPDI pin should have. Something like:

- Bootloader menu:
    - No Bootloader
    - Bootloader with UDPI pin intact (powercycle to enter bootloader)
    - Bootloader with RESET pin (DANGER - read docs first!)

SpenceKonde, 2019-12-17:
Based on above comments, it seems like this will make the core harder to use, not easier, by presenting confusing options which may or may not change behavior of the core depending on other menu options.

closed 2019-12-17 by SpenceKonde