SpenceKonde/megaTinyCore #116: parsing error

concon, 2019-11-17:
hi, I get this error when reinstalling.

"Skipping contributed index file C:\Users\lcagl\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:"

Kadah, 2019-11-17:
The full error:

```
Skipping contributed index file [...]\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected end-of-input: expected close marker for Array (start marker at [Source: (FileInputStream); line: 2, column: 15])
 at [Source: (FileInputStream); line: 1134, column: 6453]
 at [Source: (FileInputStream); line: 1133, column: 1] (through reference chain: com.fasterxml.jackson.module.mrbean.generated.cc.arduino.contributions.packages.ContributionsIndex["packages"])
```

SpenceKonde, 2019-11-17:
Will fix later today, looks like I messed up the board manager json

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-11-17:
fixed now. Reload board manager and it will work.

closed 2019-11-17 by SpenceKonde