SpenceKonde/megaTinyCore #117: eeprom attiny412

CommonGrounds, 2019-11-17:
Line 601 in boards.txt is mistake
atxy2.bootloader.SYSCFG0=0xC9
without that line eeprom eesave works fine.

SpenceKonde, 2019-11-17:
Will investigate. I thought I just fixed this, and had the setting backwards. Did I previously have it right and then change it to be wrong?

Aerospacesmith, 2019-11-18:
I haven't tried it on 1.1.x, but before that it was definitely backwards on at least the xy4 boards.

SpenceKonde, 2019-11-18:
Oh - yeah it's definitely broken on any version before 1.1.0 (and 1.1.0 is broken in general) - if you update to 1.1.0 it should work. Please let me know if it does not.

SpenceKonde, 2019-11-18:
Okay this does look to be broken in 1.1.1 as well, for at least some parts.

SpenceKonde, 2019-11-18:
Fixed, there will be an emergency 1.1.2 release to fix #118 tonight which will include this fix. Thanks for reporting. 

What I'm shocked by is that it wasn't hosing the parts entirely, as 0xC9, as far as I can tell, should have been setting the UPDI/Reset pin to act as Reset, rendering further UPDI programming impossible. Mysterious!

commit b8fc5ed4 (2019-11-18): Fix #117

eesave menu option was busted on 8, 14, and 24 pin parts.
closed 2019-11-18 by SpenceKonde