SpenceKonde/megaTinyCore #118: pins_arduino.h

concon, 2019-11-18:
I have installed 1.1.1, but ..

" In file included from C:\Users\lcagl\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.1.1\cores\arduino/UART.h:28:0,

                 from C:\Users\lcagl\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.1.1\cores\arduino/Arduino.h:178, "

There seems to be no pins_arduino.h file in UART.H, I can't compile any ino.

SpenceKonde, 2019-11-18:
Hmm, I can't seem to reproduce this issue (the pins_arduino.h that was missing in 1.1.0 is present in 1.1.1 - it's located in the variants folder for the selected board)

Before I left for work, I was just able to compile blink and analogRead serial examples for a 1614 and 412 on a portable install that had installed from board manager.

concon, 2019-11-18:
Bytheway, i find "(0)" line 43 in pins_arduino.h . I'm delete it than every thing is ok.
But Uart running in 1.0.1 does not work in 1.1.1

SpenceKonde, 2019-11-18:
Thanks - that gave me the key piece of information to fix it; it's busted in 1.1.1 in the 24-pin parts - one of the last changes I made for 1.1.1 broke it.

SpenceKonde, 2019-11-18:
Will do an emergency 1.1.2 release tonight to get the fix in.

commit 994dd137 (2019-11-18): Fix #118
closed 2019-11-18 by SpenceKonde