SpenceKonde/megaTinyCore #124: TCA0.SINGLE.PER is only 8 bit (ATtiny414)

wvmarle, 2019-12-14:
I'm experimenting with the timers, using the code below. It is supposed to make the LED blink 1 second on, 1 second off, but it doesn't.

When I attached a scope I could see the timer interrupt is actually working, but the longest period that I could get was about 13 ms, which is when I set the TCA0.SINGLE.PER register to 255. Setting it to 256 causes it to roll over, and I'm at the same frequency as a value of 1.

According to the data sheet this should be a 16 bit register. There is indeed a TCA0.SINGLE.PERH and TCA0.SINGLE.PERL defined, but the TCA0.SINGLE.PERH does not seem to have any effect.

ATtiny414
Arduino 1.8.10
megaTinyCore 1.1.2.

`const uint8_t LED = 2;

void setup() {
  TCA0.SINGLE.CTRLA = 0b00001111;                           // Prescaler clk/1024; enable the clock.
  TCA0.SINGLE.PER = 19531;                                  // 19531.25 ticks per second at 20 MHz/1024.
  bitSet(TCA0.SINGLE.INTCTRL, 0);
  pinMode(LED, OUTPUT);
  digitalWrite(LED, LOW);
}

void loop() {
}

volatile bool LEDState = false;
ISR(TCA0_OVF_vect) {
  if (LEDState) {
    digitalWrite(LED, HIGH);
    LEDState = false;
  }
  else {
    digitalWrite(LED, LOW);
    LEDState = true;
  }
  bitSet(TCA0.SINGLE.INTFLAGS, 0);
}`

wvmarle, 2019-12-14:
Turns out the timer is by default in SPLIT mode.

No idea where that happened; it goes against what the datasheet says:
> By default, the TCA is a 16-bit timer/counter. The timer/Counter has a Split mode feature that splits it into two 8-bit timer/counters with three compare channels each.

Adding the line 
`TCA0.SINGLE.CTRLD = 0;`
to setup() sets it to normal mode, the supposed default. Glad I have a scope or I'd never have been able to find this one!

SpenceKonde, 2019-12-14:
That's because the core configures it for split mode - I do that on all the parts to get the maximum number of PWM outputs - not necessary on the 8-pin parts, but very important on the parts with more pins. Be careful - timerA is used for millis() in this core (otherwise most parts wouldn't have a type B timer available for Servo or Tone), so you will also find that the overflow vector is already used and any changes you make to it's configuration will break millis() and micros().

I wonder if on the 8-pin 1-series parts we could use the type d timer, which doesn't have useful hardware pwm output pins on the 8-pin parts to do millis - the timer is a pain in the ass to work with anyway. And then we could leave the type A timer in non-split mode without the ISR used... but that's a problem for some future version.

SpenceKonde, 2019-12-14:
Oh, just saw that you're on a 414, not 412. The same applies to the type D timer there, though....

wvmarle, 2019-12-15:
Thanks for the explanation! Makes sense to do it like that if you're after more PWM control. Some documentation would be great here :-) I found out about the OVF vector, simply disabled the millis() in the menu to make that available again.
The millis() functionality on TCA0 does complicate things as TCA0 is also one of the clock sources for TCB0, and the only way to slow down TCB0 more than clk/2.
The timers have lots of functionality but as a result are indeed not easy to work with. It also took me ages to find out the names of the registers as used by the core. Another thing that needs documentation.

SpenceKonde, 2019-12-15:
Yeah - I really don't like how the type b timers don't get their own prescaler :-(

SpenceKonde, 2020-01-01:
Current plan:
Expand millis/micros tools submenu on all boards. Currently this is just enabled/disabled. This will be expanded to allow timer selection (in addition to disabling millis entirely if desires). I think the default option will be A on parts where D can do PWM, D on other parts, but with menu option to choose TCA, TCB0/1, or TCD as the timer for millis.

SpenceKonde, 2020-01-04:
This is coming along well, should be in 1.1.3-dev shortly

Also coming in 1.1.3 timeframe are options for TCB0, TCB1 (if supported)

closed 2020-01-04 by SpenceKonde